carla-simulator/carla #4332: Automating Procedural map generation with full control over assets

pavanteja295, 2021-06-22:
Hello,
I am trying to use CARLA for **procedural content generation**. Briefly, generate some traffic scenes where we have **full control** over the assets that can be placed  in the scene. Assets here are both actors and some fixed assets. I'm also focusing on a very simple use case since I feel there no direct support for this. 

Actors: Vehicles
Assets: Buildings, Length of the road until intersection. 
Rendering settings - Weather conditions. 

I'm trying to reproduce [this paper](https://arxiv.org/abs/1810.02513). Their code is not open sourced though. The authors had made substantial changes to CARLA in achieving this. But this was with Carla 0.8. Also for my use case I want to create just some **static images** where the script has full control over what elements are placed in the scene. For each element in the scene we configure the location to place it and also if we would want to place it or not and blueprint of the asset.

For actors one can easily use CARLA python API but for assets there is no support yet.

Firstly, with current CARLA API, we load pre-existing maps or some customised map manually created from unreal editor. So fixed assets cannot be changed. [Toggle Environment Objects and Map sublevels](https://carla.org/2020/12/22/release-0.9.11/) might help to a certain extent but I would also like to have building type.

I also looked the recent scenic support to CARLA. However, one can only modify actors with scenic extension. The authors of the paper mentioned above were successful to use CARLA for fully controlled procedural scene generation I'm not sure how.I want to use the full support for weather conditions, different blueprints existing in CARLA but also want to have full control over the assets mentioned above.  

I am new to Unreal editor and I would also love to opensource this as procedural content generation is quite common these days and having something like this in CARLA would be very helpful. 

Let me know  if someone has already done this and  let me know any suggestions in achieving this. 

Thank you so much for your time.

ish-101, 2021-06-23:
Hello,

I am also working on a project using CARLA that requires automated procedural map generation. I am trying to mutate assets on the map, and also the road structure, if possible. Assets could include core traffic objects like crosswalk, stop signs, or traffic signals. So far I have only seen RoadRunner and CARLA Map Editor do that, and they're both GUI-based. I would really appreciate if someone who knows more about CARLA than me could help me get in the right direction.

Thank you!

algababr, 2021-07-12:
Hi, this is Roel member of the CARLA Team.

This sounds quite interesting! It is true that right now we don't have full support for generating custom maps for CARLA. We have some functionalities that help us get objects from our default custom maps and information about them, but nothing implemented for generating them procedurally. Right now the team is focusing in the next release of CARLA which will be quite soon and since this is a quite big feature this would take some time to prepare. But we will discuss this as soon as we are able to. I'll update this issue once we have green light for doing it.

pavanteja295, 2021-08-15:
Sure, thanks for getting back.

closed 2023-06-02 by stale[bot]