carla-simulator/carla #4333: Getting walker paths and interaction with Unreal Blueprints

mfahrl, 2021-06-24:
Hello! 
I generated a small custom map and populated it with walkers that are controlled by the WalkerAIController. However, to make the pedestrian behavior a little more realistic, I am currently trying to trigger animations based on future behaviour, such as triggering a Look-Left-and-Right animation, right before the pedestrian is crossing the street. 

To do so, I need to know when a pedestrian is going to cross, so I am trying to get the path the Navigation has planned for it.  If I read the Code correctly the pedestrian paths are generated in the Navigation.cpp, but while the WalkerAIController has a go_to_location() method it is lacking an equivalent get function. Can you tell me if there is a possibility already to retrieve the pedestrian path or would I have to write a get function myself?

Also, to trigger anmiations in Unreal from the PythonAPI I am currently using a workaround of sending a jump command and playing a custom animation instead of a jump animation. Can you tell me if there is a better way to do this? Is it possible to trigger events or adjust variables in Unreal blueprints from within the PythonAPI? I'd be grateful for any hint here.

The use case here is to generate a diverse dataset of realistic pedestrian behavior at various positions in the map, captured from a camera attached to a car - Do you think using the ScenarioRunner would be a better approach in general here or is it's main purpose to play the same scenarios over and over again? 

Thanks a lot for any help and hints!

closed 2021-07-08 by mfahrl

mfahrl, 2021-07-08:
I added a custom get_path method for the WalkerAIController and a play_animation method for Walkers that uses Unreals PlayAnimMontage() function. Closed the issue.

LucasFKobernic, 2021-07-09:
Hi @mfahrl i trying to do the same as you. Would you like to discuss about the solution? Would you share your play_animation fucntion?

mfahrl, 2021-07-12:
Hi @LucasFKobernic to add the functions I followed the procedure described in issues #2143 and #1615. I basically checked how the  walker function "apply_control" is called and followed the function calls from client to server and added a play_animation function in each file that just passes on a string. The string is meant to be the path to the animation to be played.
Then, in the [WalkerController.cpp](https://github.com/carla-simulator/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Walker/WalkerController.cpp) I overloaded the ControlTickVisitor, to load and play the animation:

`void AWalkerController::ControlTickVisitor::operator()(const std::string &AnimPath)
{
	FString Path(AnimPath.c_str());
	auto *Character = Controller->GetCharacter();
	if (Character != nullptr)
	{
		UAnimMontage * AnimMontage = LoadObject<UAnimMontage>(NULL, *Path);
		if (AnimMontage != nullptr) {
			Character->PlayAnimMontage(AnimMontage);
		}
		else {
			GEngine->AddOnScreenDebugMessage(-1, 15.0f, FColor::Blue, TEXT("Failed to Load AnimMontage"));
		}
	}
}`

LucasFKobernic, 2021-07-12:
Hi @mfahrl,

thanks for the answer, i will try it!

hiber-wang, 2023-06-13:
@mfahrl Hello, I trying to get the walker paths as you. Would you like to talk about the solution? Would you share your get_path fucntion, thank you

cucdengjunli, 2025-04-13:
厉害