carla-simulator/carla #4344: cannot import name 'VehicleLightState' from 'carla' (unknown location)

Namareq33, 2021-06-28:
hi everybody 
i need help with carlasimulater, i use v0.9.11 and pycharm ide , 
when i run the first exaple of spwan_npc this error appear
cannot import name 'VehicleLightState' from 'carla' (unknown location)
i use python 3.8
with conda enviroment

mkhansenbot, 2021-10-05:
I'm running into the same issue. I've tried running python interactively and tried both the 0.9.5 carla python package and the 0.9.12 package. When I run the 0.9.12, I get a core dump due to an API mismatch:
```
carla@ip-172-31-41-42:~$ python3 PythonAPI/examples/spawn_npc.py 
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 0.9.12 
WARNING: Simulator API version  = 0.9.11 
Segmentation fault (core dumped)
```
When I install 0.9.5 (there doesn't seem to be a 0.9.6-11 in PyPi) I get the same error as @corkyw10 above. When I run python3 inside the container, I can see that API isn't there.
```
>>> import carla
>>> from carla import VehicleLightState as vls
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'VehicleLightState'
>>> carla.
carla.Actor(                   carla.GnssEvent(               carla.Rotation(                carla.Walker(
carla.ActorAttribute(          carla.GnssSensor(              carla.Sensor(                  carla.WalkerControl(
carla.ActorAttributeType(      carla.Image(                   carla.SensorData(              carla.Waypoint(
carla.ActorBlueprint(          carla.LaneChange(              carla.ServerSideSensor(        carla.WeatherParameters(
carla.ActorList(               carla.LaneInvasionEvent(       carla.Timestamp(               carla.WheelPhysicsControl(
carla.BlueprintLibrary(        carla.LaneInvasionSensor(      carla.TrafficLight(            carla.World(
carla.BoundingBox(             carla.LaneMarking(             carla.TrafficLightState(       carla.WorldSettings(
carla.Client(                  carla.LaneMarkingColor(        carla.TrafficSign(             carla.command
carla.ClientSideSensor(        carla.LaneMarkingType(         carla.Transform(               carla.libcarla
carla.CollisionEvent(          carla.LaneType(                carla.Vector2D(                carla.vector_of_ints(
carla.Color(                   carla.LidarMeasurement(        carla.Vector3D(                carla.vector_of_vector2D(
carla.ColorConverter(          carla.Location(                carla.Vehicle(                 carla.vector_of_wheels(
carla.DebugHelper(             carla.Map(                     carla.VehicleControl(          
carla.GeoLocation(             carla.ObstacleDetectionEvent(  carla.VehiclePhysicsControl(   
>>> carla.Vehicle
carla.Vehicle(                carla.VehicleControl(         carla.VehiclePhysicsControl(  
```
Here's the Dockerfile I'm using to build carla 0.9.11:
```
FROM carlasim/carla:0.9.11

USER root

# install dependencies
RUN apt update && \
      apt install -y python3-pip \
      libjpeg-dev \
      libtiff5-dev \
      libomp-dev \
      fontconfig

# fix ALSA errors
RUN echo pcm.!default { type plug slave.pcm "null" } >> /etc/asound.conf

# install NICE DCV (for RoboMaker)
RUN apt update -y && apt upgrade -y && apt install -y wget pgp
RUN wget https://d1uj6qtbmh3dt5.cloudfront.net/NICE-GPG-KEY
RUN gpg --import NICE-GPG-KEY
RUN wget https://d1uj6qtbmh3dt5.cloudfront.net/2021.1/Servers/nice-dcv-2021.1-10598-ubuntu1804-x86_64.tgz
RUN tar -xvzf nice-dcv-2021.1-10598-ubuntu1804-x86_64.tgz
RUN apt update && apt install -y ./nice-dcv-2021.1-10598-ubuntu1804-x86_64/nice-dcv-gl_2021.1.937-1_amd64.ubuntu1804.deb ./nice-dcv-2021.1-10598-ubuntu1804-x86_64/nice-dcv-gltest_2021.1.275-1_amd64.ubuntu1804.deb

# install opengl 
RUN apt update && apt install -y libglfw3 libglfw3-dev

# install xterm
RUN apt update && apt install -y xterm

# run as user carla
USER carla

# install example dependencies
RUN python3 -m pip install -U pip
RUN cd ~/PythonAPI/examples && python3 -m pip install -r requirements.txt
RUN python3 -m pip install carla==0.9.5
```
Does the example work on 0.9.11? I can't run 0.9.12 because it's Vulkan only and I need to run OpenGL graphics.

mkhansenbot, 2021-10-06:
Update: I resolved this issue by setting the PYTHONPATH env var to point to the carla egg file in my Dockerfile:
`ENV PYTHONPATH=/home/carla/PythonAPI/carla/dist/carla-0.9.11-py3.7-linux-x86_64.egg`

I tried `make PythonAPI` to install it, I also tried to install it using `python3 -m pip install` neither of those worked. 

**Is there a way to install it instead of using the PYTHONPATH env var?**

mkhansenbot, 2021-10-07:
Are you running from a docker container? If so, just build and run my Dockerfile above, but replace the last line with the `ENV` statement above.

If not running from a container, then use `export PYTHONPATH=` before running the `spawn_npc.py` script

Let me know if that works.

closed 2023-06-02 by stale[bot]