carla-simulator/carla #4345: Is there a way to set the timeout parameter for traffic_manager.set_synchronous_mode?

aaronh65, 2021-06-28:
I'm facing an issue in a CARLA project that I believe could be resolved if I could change the timeout parameter associated with `traffic_manager.set_synchronous_mode()` as described in the C++ docs [here](http://carla.org/Doxygen/html/df/d5a/classcarla_1_1traffic__manager_1_1TrafficManager.html#afe3ae045966f9fed31a8d4d9c11554cc). However, I see no way to set this parameter either through the `traffic_manager` Python API or from `CarlaUE4.sh` command line arguments.

Is there a way to do this?

jackbart94, 2021-06-29:
You can change it in the C++ API. Go to ``Parameters.cpp`` and change ``  synchronous_time_out = std::chrono::duration<int, std::milli>(10);`` to whatever value you want. Then, do ``make LibCarla PythonAPI``.

qhaas, 2021-06-29:
IMO, the timeout dynamics with rpclib and/or boost::asio need to be re-imagined, per Issue #3707

aaronh65, 2021-06-29:
Thanks for the response, I haven't tried compiling from source ever - I've just been using the precompiled stuff. Is there a way to specify the set_synchronous_mode timeout argument without recompiling?

jackbart94, 2021-07-12:
Right now, not really.

closed 2021-11-18 by jackbart94