carla-simulator/carla #4346: Change STOP and CROSSWALK segmentation color

protanjung, 2021-06-29:
Hi, i want to get "road lane only" semantic segmentation of carla. But i fount out that road marking and road lines are in the same color, e.g STOP and Crosswalk. Is there a way for me to change the color of STOP and Crosswalk? So in the end, the semantic segmentation has different color between road lanes and road markings.
[image]
[image]

bhuvanofc, 2021-11-17:
@Axel1092 Did u find any solution to this problem? If so please let me know i have a similar problem too. Would be really helpful. Thanks

bhuvanofc, 2021-11-17:
@Axel1092 @protanjung Did u find any solution to this problem? If so please let me know i have a similar problem too. Would be really helpful. Thanks

protanjung, 2021-11-17:
Sorry, but I don't find any workaround yet. Perhaps we have to modify and build carla from source (?)

protanjung, 2022-04-26:
Is there anyone with this same issue? Hmm, i am still wondering around to solve this. Hahaha.

Blyron, 2024-11-07:
You can use this tuto https://carla.readthedocs.io/en/latest/tuto_D_create_semantic_tags/