carla-simulator/carla #4347: Waypointing issues with Town03 roundabout

seawee1, 2021-06-29:
I'm trying to calculate a set waypoints between two custom road positions using the GlobalRoutePlanner, i.e. via calling the `set_destination` function of the BehaviourAgent. Somehow the waypoints seem to be broken for the roundabout (see attached picture)? I'm using Carla 0.9.11.

[image]

seawee1, 2021-06-30:
Okay, so we found a workaround: We increase the _sampling_resolution of the BehaviorAgent from 4.5 m to 0.5 m and downsample the waypoints afterwards.

glopezdiest, 2021-07-13:
Hey @seawee1. That is indeed the case. The global planner uses the topology of the road to create the graph. This works fine but the roundabout lanes are pretty hectic and with a resolution of 4.5m, the resulting route is weird.

This is quite an old feature and we are improving it right now. When it's ready, this problem won't be there anymore

closed 2021-07-13 by glopezdiest

zengxianming, 2022-07-12:
I got the same bug again. I set my sample resolution to 0.1m, it still doesn't work. I use CARLA 0.9.12. Here is my test code
```python 
from agents.navigation.global_route_planner import GlobalRoutePlanner
import carla
import time

client = carla.Client("localhost", 2000)
client.set_timeout(2.0)

world = client.get_world()
map = world.get_map()

sampling_resolution = 0,1
global_planner = GlobalRoutePlanner(map, sampling_resolution)

start_location = carla.Location(x=52.61, y=-7.84, z=1.84)
end_location = carla.Location(carla.Location(x=-84.81,y=27.02,z=1.84))
res = global_planner.trace_route(start_location, end_location)

while True:
    for wpt, _ in res:
        world.debug.draw_point(wpt.transform.location, size=0.5, life_time=1.0)
    time.sleep(1)
```
[image]