carla-simulator/carla #4350: Vague Segmentation fault with carla.walkeraicontroller

qhaas, 2021-06-29:
I believe an exception isn't making its way up from the depths of the C++ client library into a form that can be caught in the python3 side.  This leads to difficulty in debugging and this problem might arise under other circumstances than the one described here.

Consider this minimal reproducible example that ends simply with the python script exiting with only a `Segmentation fault (core dumped)` in CARLA 0.9.11 in GNU/Linux:
```
#!/bin/python3

import carla
import traceback

try:

    # setup client/server
    client = carla.Client('localhost', 2000)
    world = client.load_world('Town03')
    settings = world.get_settings()
    settings.fixed_delta_seconds = 1./20
    settings.synchronous_mode = True
    world.apply_settings(settings)
    world.tick()

    spawnPoint = carla.Transform()
    spawnPoint.location = world.get_random_location_from_navigation()
    walkerActor = world.spawn_actor(world.get_blueprint_library().find('walker.pedestrian.0001'), spawnPoint)
    walkerControllerActor = world.spawn_actor(world.get_blueprint_library().find('controller.ai.walker'), carla.Transform(), walkerActor)
    #world.tick()
    walkerControllerActor.start()
    walkerControllerActor.go_to_location(world.get_random_location_from_navigation())
    world.tick()
    settings.fixed_delta_seconds = None
    settings.synchronous_mode = False
    world.apply_settings(settings)
    world.wait_for_tick()

except Exception as err:
    print(traceback.format_exc())
```

The code above is 'wrong' because the `world.tick()` between walkeraicontroller creation and start, as [recommended in the examples](https://github.com/carla-simulator/carla/blob/0.9.11/PythonAPI/examples/spawn_npc.py#L250), is commented out.  Uncomment that line and it works fine.  Curiously, it works with `Town01` even when that tick is commented out.

qhaas, 2021-08-15:
Issue still exists and behaves the same way in 0.9.12

adwiii, 2023-06-29:
I am hitting what I think is the same issue in `0.9.14`, though it doesn't appear to be fixed by ensuring that there is a world tick between the spawning of the walker and adding the controller. To be clear, the segfault happens when `go_to_location` is called - the calls to `spawn_actor` and `start` are successful. The script I am testing with uses a seeded random to spawn actors. It does not segfault on the first actor, and how many must be added before faulting is consistent between runs with the same seed, but may change with different seeds. It appears to be dependent on the location parameter sent, though this also may be dependent on the location of the actor.

PabloVD, 2024-11-22:
I have run the code above in the Leaderboard CARLA package (based on `0.9.14`) in ubuntu 20 and I couldn't reproduce the segmentation fault error.

Either if I uncomment the `world.tick()` or not, the python script runs normally without crashing. However, the `world.tick()` between the spawn of `walkerControllerActor` and the `start` is required to get the pedestrian walking. Otherwise, the pedestrian does not start the walking animation. This is consistent with the [example commented above](https://github.com/carla-simulator/carla/blob/0.9.11/PythonAPI/examples/spawn_npc.py#L250).

This could be related to [this issue](https://github.com/carla-simulator/carla/issues/8267).

Expected behaviour: when `world.tick()` before `walkerControllerActor.start()` is uncommented, start walking animation.
Current behaviour: same.

closed 2025-08-26 by Blyron