carla-simulator/carla #4363: RuntimeError: Unknown error while trying to spawn actor

dhiegomaga, 2021-07-02:
**Carla version: 0.9.11**

I tried following the tutorial to add Props to carla [here](https://carla.readthedocs.io/en/latest/tuto_A_add_props/#ingestion-in-a-carla-package). More specifically, I followed the "Ingestion in a build from source", i.e., I have a locally compiled Carla, which I run through Unreal Engine. 

Added the files to the `Import/` folder exactly like the tutorial described: 

carla/
--Import/
----Package01/
----|--Package01.json
----|--Props/
----|--|--Ball/
----|--|--|--Ball.fbx

Package01.json:
```json
{
    "maps": [
    ],
    "props": [
      {
        "name": "Football",
        "size": "small",
        "source": "./Props/Ball/Ball.fbx",
        "tag": "Other"
      }
    ]
  }
```

Used this [object](https://free3d.com/3d-model/football-ball--64059.html). 

I tried spawning it (`world.spawn_actor()...`), even though it finds it on the Blueprint list (with `blueprint_library.filter('static.prop.football')[0]`), I get an error:

```bash
RuntimeError: Unknown error while trying to spawn actor
```

dhiegomaga, 2021-07-08:
I noticed the error happens when the asset doesn't appear as "static mesh" inside Unreal, (e.g, "Skeletal Mesh"). But there is nowhere explaining how to export from Blender to Carla an .fbx file that is a "static mesh".

GustavoSilvera, 2021-09-21:
Hi, I was facing the exact same issue when trying to spawn new props and I found a valid workaround (for `props` at least). 

If you look at `carla/Unreal/CarlaUE4/Content/Carla/Config/Default.Package.json` this file defines a large list of valid props and their static mesh source destinations as spawnable props in CarlaUE4. 

With this, I was able to add a `new_prop` after the last one (`Calibrator` in 0.9.11) that looked like:
```json
{
	"name": "new_prop",
        "path": "/Game/Carla/Static/Static/SM_new_prop.SM_new_prop",
	"size": "Medium" # or whatever size estimation you'd like
}
```

Of course ensure you correctly have valid static mesh path file (note that the file does have a `.uasset` filetype, but in the json you should still denote it as `SM_new_prop.SM_new_prop` instead of `SM_new_prop.uasset`). 

Then after a quick relaunch of the editor you should be able to now spawn your `new_prop` from the blueprint library as follows:
```python
bp = list(blueprint_library.filter("static.prop.new_prop"))[0]
transform = world.get_map().get_spawn_points()[0] # or choose any other spawn point
world.spawn_actor(bp, transform) # should succeed with no errors
```

Note that I've only tested this in Town03, and interestingly some of the additional (opt) maps (in 0.9.11) have their own `.Package.json` files which you might need to modify similarly to reflect in those maps.

ChengHuang-CH, 2021-10-04:
I also experienced the same issue and solved it after several tests.Thanks for @GustavoSilvera mentioning the existence of the file *.Package.json, I was able to find out the problem, which is caused by the naming rule  "Override FullName" of UE4 while importing the .fbx file.

For example, my static mesh name in Blender is "MeshName", and the exported .fbx file is "FbxName.fbx".  After importing all files into carla environment following offical instructions, I find that the static mesh name became "FbxName_MeshName" instead of the expected "FbxName" inside the UE4 content folder. My solution is to revise the  "FbxName.FbxName" to "FbxName_MeshName.FbxName_MeshName"  in the generated *.Package.json file after checking the real name.

All in all, it is important  to check the real name of the imported static mesh inside the content folder of UE4, and see if it is the same as in  (your_map_name).Package.json file.  

(Besides, I cannot just import new props by filling the "props" content and leave the "maps" content empty in the json file, it is necessary for me to add new map files ( .fbx and .xodr) at the same time, otherwise, "import_assets_from_json_list   if "tile_size" in maps[0]: IndexError: list index out of range" issue will break the process of importing new props. And I didn't figure out the reason behind)

crabiner, 2021-10-04:
I also faced the same issue in **Carla version: 0.9.12** modified the  generated `*.Package.json` file as you said to` "FbxName_MeshName.FbxName_MeshName"` and it fixed it

crabiner, 2021-10-04:
> (Besides, I cannot just import new props by filling the` "props"` content and leave the` "maps"` content empty in the json file, it is necessary for me to add new map files ( .fbx and .xodr) at the [...]

I don't have any "maps" in json and to overcome the error you see try editing Util/BuildTools/Import.py Util/BuildTools/Import.py script
replace two instances of `if "tile_size" in maps[0]:` with `if len(maps) > 0 and "tile_size" in maps[0]:`
I also had to comment out the line with ` copy_roadpainter_config_files(args.package)`

bernatx, 2021-10-05:
Hi folks,
You are right, it seems there are some problems when importing props without a map. I'm fixing that.

The problem with the nomenclature adding the FBX name before the asset is because now you can import maps in tiles, and then we need to avoid conflicts between assets with the same name from different tiles, that is why we add the FBX name as a prefix.

Regards

chengguizi, 2021-10-30:
I also encountered the issue that `make pacakge ARGS="--packages=<package>"` does not cook props correctly in `0.9.12`. This results in unable to spawn props when being imported to packaged Carla.

The fix for me is to add after this [line](https://github.com/carla-simulator/carla/blob/0.9.12/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Commandlet/PrepareAssetsForCookingCommandlet.cpp#L514)

`MapPathDataLinux.Append(PropsMapPath + TEXT("/PropsMap"));`

I believe the cause is the missing PropsMap when cooking for Linux version. Hope it helps!

YHao29, 2023-04-14:
@GustavoSilvera hello, I'm so excited that you give me some hope. But I still meet some problem about using my own prop.
### Prepares
I use the **docker version of carla 0.9.13**  and I have design a new prop then import to the docker following this [guide.](https://carla.readthedocs.io/en/latest/tuto_A_add_props/) .
After I run the ` ./ImportAssets.sh`, I can see a ` Package01` has been created in the filefolder :
[image]

### Problem
But when I want to spawn it to the world, I can not find it in the blueprint. And I also don't know how to use it like other props (` 'static.prop.bench01'`). 

### Question
Could you teach me that how to use the props I just import to the **carla docker**? thanks for your time

2019211753, 2023-08-16:
In Carla_0.9.14, it's still a problem. and i solved it by @GustavoSilvera's solution. However, my path is the absolutely path instead of "/Game/..."

lsangreg, 2023-09-21:
Hello, I followed your instruction and i was able to obtain a correct package from ue4 to carla, with .uexp and .uasset files. The problem now is that  I still get the error 
`RuntimeError: Unknown error while trying to spawn actor`
When i try to spawn the actor!
I also added my own lines in Default.Packages.json of the simulator /CarlaUE4/Content/Carla/Config
``` 
		        {
            "name": "Pallet",
            "path": "/Game/Pallet/Static/Other/Pallet/pallet.pallet",
            "size": "medium"
        }```

Croquembouche, 2024-04-30:
This is still an issue.

Blyron, 2024-05-17:
Which is actually the issue? 
If you have the uasset in content folder then you add properly the path. The spawn works.

lsangreg, 2024-05-17:
@Croquembouche  check this https://github.com/carla-simulator/carla/issues/6789

ShaharShakiUnreal, 2024-05-22:
Hi, I also have this problem.
I follow the guides here, but still get the "Unknown error while trying to spawn actor".
I was added this guide https://carla.readthedocs.io/en/latest/tuto_A_add_props/#ingestion-in-a-build-from-source
and added the fbx file: Import/Package01/Props/bigrashcan/bigtrashcan.fbx
also added the Package01 json with this content:
{
  "maps": [
  ],
  "props": [
    {
      "name": "bigtrashcan",
      "size": "medium",
      "source": "./Props/bigtrashcan/bigtrashcan.bigtrashcan",
      "tag": "TrashCan"
    }
  ]
}

also updated the Default.Package.json with this:
		{
			"name": "bigtrashcan",
			"path": "/Game/Carla/Static/Static/Trashcans/bigtrashcan_Wheels_Auto1.bigtrashcan_Wheels_Auto1",
			"size": "Medium"
		}
		
		and still I can see the prop on the blueprint library list, but when spawn it getting the error I mentioned above.
		can someone please help? thanks.

Blyron, 2024-05-22:
How are you trying to spawn it? example code please.
Have you tried to get list of all props and see if your big trash can is there?

ShaharShakiUnreal, 2024-05-22:
> How are you trying to spawn it? example code please. Have you tried to get list of all props and see if your big trash can is there?

Hi, Thanks for you r quick reply.
[image]
as you see I get the big trash can in the props list. But inside the "try" block when I get to the world.spawn_actor I get the error I mentioned above.
Thanks

Blyron, 2024-05-22:
Print in console of static.prop and see what happens
It may be there

ShaharShakiUnreal, 2024-05-22:
> Print in console of static.prop and see what happens It may be there

do you mean like that?:
[image]

Blyron, 2024-05-22:
Yes, your bigtrashcan is there as expected, so your method to get it is wrong.

ShaharShakiUnreal, 2024-05-22:
> Yes, your bigtrashcan is there as expected, so your method to get it is wrong. Try something like prop = world.get_blueprint_library().filter("static.prop.bigtrashcan") spawn_point = random.choice(spawn_points) spawn_point.location.z += z
> 
> SpawnActor(prop, spawn_point)

But I manage to spawn vehicles without problems:
car_actor = world.spawn_actor(car_blueprint, spawn_point)

when I do the same to the trash can it failed.
I think the problem might be in the json or naming convention of the files.

Blyron, 2024-05-22:
I think is related with you try to get the proper blueprint

ShaharShakiUnreal, 2024-05-22:
> I think is related with you try to get the proper blueprint

I'm not sure I get it. I'm trying to spawn it as you said, unfortunately it is not working

Blyron, 2024-05-22:
prop = world.get_blueprint_library().filter("static.prop.bigtrashcan")
spawn_point = carla.Transform()
spawn_point.location.z = 200.0
spawn_point.rotation.roll = 0.0
spawn_point.rotation.pitch = 0.0
world.try_spawn_actor(prop, spawn_point)

Is this working?

Blyron, 2024-05-22:
I mean the fucntioanlity is working and tested by different people something must be going on wrong on your code

ShaharShakiUnreal, 2024-05-22:
> prop = world.get_blueprint_library().filter("static.prop.bigtrashcan") spawn_point = carla.Transform() spawn_point.location.z = 200.0 spawn_point.rotation.roll = 0.0 spawn_point.rotation.pitch = 0.0 world.try_spawn_actor(prop, spawn_point)
> 
> Is this working?
Getting this error:

Python argument types in
    World.try_spawn_actor(World, BlueprintLibrary, Transform)
did not match C++ signature:
    try_spawn_actor(class carla::client::World {lvalue}, class carla::client::ActorBlueprint blueprint, class carla::geom::Transform transform, class carla::client::Actor * __ptr64 attach_to=None, enum carla::rpc::AttachmentType attachment_type=carla.libcarla.AttachmentType.Rigid)

Blyron, 2024-05-22:
```
client = carla.Client(args.host, args.port)
client.set_timeout(2000.0)

world = client.get_world()
prop = world.get_blueprint_library().filter("static.prop.bigtrashcan")
spawn_point = carla.Transform()
spawn_point.location.z = 200.0
spawn_point.rotation.roll = 0.0
spawn_point.rotation.pitch = 0.0
world.try_spawn_actor(prop, spawn_point)
```

ShaharShakiUnreal, 2024-05-22:
> ```
> client = carla.Client(args.host, args.port)
> client.set_timeout(2000.0)
>
> world = client.get_world()
> prop = world.get_blueprint_library().filter("static.prop.bigtrashcan")
> spawn_point = [...]

I guess you meant:
prop = world.get_blueprint_library().filter("static.prop.bigtrashcan")
                spawn_point = carla.Transform()
                spawn_point.location.z = 200.0
                spawn_point.rotation.roll = 0.0
                spawn_point.rotation.pitch = 0.0
                world.try_spawn_actor(**prop[0]**, spawn_point)

and that not working either. Its not crashing but the object is not spawned.

Croquembouche, 2024-05-29:
> Which is actually the issue? If you have the uasset in content folder then you add properly the path. The spawn works.

Hi, I managed to import some props, but some refused to load. These props are from the Unreal Engine and CARLA blueprint. I am trying to package the props in the source version and import them into the prebuilt version.

I will try again later and update this issue with screenshots.

astralmaster, 2025-04-12:
I spent 40 hours on this error. The props would spawn in Carla build from source version but the packaged version would not even copy the asset files to the package build directory. Obviously found this thread by googling, and since I exist, I am sure someone else will eventually stumble upon the same unique case too. Nothing upper mentioned helped me.

When you run the `make package`  command, the code in [PrepareAssetsForCookingCommandlet.cpp](Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Commandlet/PrepareAssetsForCookingCommandlet.cpp) validates asset files inside your package before putting them on map and generating the needed files. One such validation involves checking if the asset file name [contains](https://github.com/carla-simulator/carla/blob/d7b45c1e159e6d13296f7a3a4e8b13e6c2d62c18/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Commandlet/PrepareAssetsForCookingCommandlet.cpp#L28) strings "light" or "sign". And guess what, every single one of my props contains "sign" in it, due to having a word "signature" as part of them. So the validation would not pass, and the assets were not added to the map, and consequently not exported in the package. 
Dare I venture a guess, the method ValidateStaticMesh() could have been assembled using a more sound logic.

closed 2026-11-12 by Madecu