carla-simulator/carla #4365: How to develop in docker?

EthanCodesss, 2021-07-05:
Hello! I installed carla on the host. And I want to run carla on the host，run ros in docker, use ros bridge for development.
Except docker, ros python3 etc are installed in docker.
But I get the message "No module named carla". How can I set up pythonApi in docker. And make it work with the host?
I use : Ubuntu 18.04, docker 19.03, and Carla 0.9.11.
Thanks for all help!!

qhaas, 2021-07-06:
CARLA ships with an egg that can be installed in a similar python environment using `easy_install`, see: https://github.com/carla-simulator/carla/issues/4287#issuecomment-858804822

EthanCodesss, 2021-07-07:
> CARLA ships with an egg that can be installed in a similar python environment using `easy_install`, see: [#4287 (comment)](https://github.com/carla-simulator/carla/issues/4287#issuecomment-858804822)
Thanks!!  I'll try.  The problem is closed.

closed 2021-07-07 by EthanCodesss