carla-simulator/carla #4366: lidar_to_camera.py bug

helloworldunicon, 2021-07-05:
lidar_to_camera.py  
 
The output radar map, why is there a map of radar signal behind

DSantosO, 2021-07-14:
Hello @helloworldunicon,
What do you mean? There are no radars in that script...

helloworldunicon, 2021-07-14:
已经解决了 ，是调参测试的时候，每分钟的点数太少，会导致地图上某些位置 生成不了雷达数据，不过雷达的每分钟点数我调到1000000，会不会非常影响程序的效率。

DSantosO, 2021-08-04:
Hello @helloworldunicon,
Sorry about the late response.
I am glad that you solved it. 
About your question: yes, increasing the number of points affects greatly to the performance. 
This is difficult to say numbers because depends on the scenario and your hardware but it general duplicate the number of points per second duplicates the work so expect a drop in performance accordantly.

closed 2021-09-03 by DSantosO

helloworldunicon, 2021-10-20:
&nbsp;help？thankyou。