carla-simulator/carla #4367: Carla error ros_compatibility

spanthr, 2021-07-05:
File "/home/shaurya/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/actor_factory.py", line 33, in <module>
    from carla_ros_bridge.marker_sensor import MarkerSensor
  File "/home/shaurya/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/marker_sensor.py", line 29, in <module>
    carla.CityObjectLabel.Buildings: ColorRGBA(r=70.0/255.0, g=70.0/255.0, b=70.0/255.0, a=0.8),
AttributeError: 'module' object has no attribute 'CityObjectLabel'
================================================================================REQUIRED process [carla_ros_bridge-2] has died!
process has died [pid 9094, exit code 1, cmd /home/shaurya/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py __name:=carla_ros_bridge __log:=/home/shaurya/.ros/log/36e001b0-dd7b-11eb-a982-94e6f7a668aa/carla_ros_bridge-2.log].
log file: /home/shaurya/.ros/log/36e001b0-dd7b-11eb-a982-94e6f7a668aa/carla_ros_bridge-2*.log
Initiating shutdown!
================================================================================
[carla_manual_control_ego_vehicle-5] killing on exit
[set_initial_pose_ego_vehicle-4] killing on exit
[carla_spawn_objects_shauryapc_9067_1718727163877865638-3] killing on exit
[carla_ros_bridge-2] killing on exit
Traceback (most recent call last):
  File "/home/shaurya/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_spawn_objects/src/carla_spawn_objects/carla_spawn_objects.py", line 378, in <module>
    main()
  File "/home/shaurya/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_spawn_objects/src/carla_spawn_objects/carla_spawn_objects.py", line 356, in main
    spawn_objects_node = CarlaSpawnObjects()
  File "/home/shaurya/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_spawn_objects/src/carla_spawn_objects/carla_spawn_objects.py", line 56, in __init__
    self.spawn_object_service = self.new_client(SpawnObject, "/carla/spawn_object")
  File "/home/shaurya/carla-ros-bridge/catkin_ws/src/ros-bridge/ros_compatibility/src/ros_compatibility/node.py", line 89, in new_client
    raise ROSException(e)
ros_compatibility.exceptions.ROSException: rospy shutdown
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

corkyw10, 2021-07-12:
@spanthr could you provide more context on your error please?

zacgeiser, 2021-12-15:
Does any one found the solution for this error?

zacgeiser, 2021-12-15:
or a good tutorial to install and launch Carla ros autoware without facing any error?

jiangqianpeng, 2023-12-06:
add the path into .bashrc
gedit ~/.bashrc
and add your path
export PYTHONPATH=SPYTHONPATH:~/carla/PythonApI/carla/dist/carla-0.9.13-py2.7-linux-x86-64.egg
make sure the python is version 2.7

Blyron, 2026-11-12:
Thanks everyone for the reports, and sorry this sat for so long.

**Why it fails in 0.9.x.** `carla.CityObjectLabel` was added to the Python API in CARLA 0.9.10. `AttributeError: 'module' object has no attribute 'CityObjectLabel'` means the ROS bridge is importing an older `carla` module than the one it was written for. That happens when the bridge release doesn't match the CARLA version, or when `PYTHONPATH` points at an old egg (or at an egg built for another Python version). To fix it on the UE4 line:

- Use the ros-bridge release that matches your CARLA server. The bridge's own docs list the compatible pairs: https://carla.readthedocs.io/projects/ros-bridge/en/latest/
- Make sure the `carla` package the bridge imports comes from that same CARLA release. Use Python 2.7 for ROS Melodic and Python 3 for Noetic/ROS 2. Check with `python -c "import carla; print(carla.__file__)"`, and fix that before re-launching the bridge.

**Going forward.** We are migrating to `ue58-dev`, which becomes **CARLA 1.0** (Unreal Engine 5.8). That version doesn't need the ROS bridge any more. CARLA speaks **ROS 2 natively** from inside the simulator, so there is no separate bridge process and no Python-version matching to get right. Start the server with `-ros2`, and sensors publish and the ego vehicle subscribes to control commands directly. You can choose Fast DDS, Cyclone DDS or Zenoh as the middleware.

- ROS 2 native interface: https://carla-ue5.readthedocs.io/en/latest/ros2_native/
- Published sensors, topic names and QoS: https://carla-ue5.readthedocs.io/en/latest/ros2_native_sensors/

@zacgeiser, about a tutorial for running CARLA with Autoware: 1.0 ships an Autoware integration on top of this interface, with no bridge. It covers the classical stack and the VAD end-to-end mode, an installer and a launcher, and the map preparation tools:

- Autoware: https://carla-ue5.readthedocs.io/en/latest/autoware/

If you hit problems with the native ROS 2 interface on `ue58-dev`, please open a new issue with your setup.