carla-simulator/carla #4369: No determinism across different time step settings in 0.9.11.

brscholz, 2021-07-06:
Hello everyone,

I have a problem with determinism of the exact same simulation case run at different time steps under version 0.9.11.

Description of my use case:

- 1 ego vehicle, 2 target vehicles, all operated by traffic manager on a small custom town course map
- Simulation 1: sync mode, fixed time step 0.027 s, max phys substep 0.001 s, max substeps 26, 2 rgb cameras, random seed 42
- Simulation 2: sync mode, fixed time step 0.005 s, max phys substep 0.001 s, max substeps 4, 3 custom sensors, random seed 42
- Initial phase of vehicles dropping until their velocity magnitude goes below 1e-3 m/s, then traffic manager speed target is set to 20 km/h

I need to get both simulations exactly sync to be able to merge the images and the sensor data into one data stream later on. If I want to merge them into one simulation, which was my initial try, I have to go with a fixed time step of 1 ms which slows down simulation speed by an inacceptable amount.

Unfortunately even with the physics substep forced to 1 ms in both simulations, I get different vehicle accelerations, thus speeds, thus positions of the vehicles. The initial acceleration in the 5 ms case is about 50% higher than in the 27 ms case. I suppose this is due to the traffic managers velocity controller.

My problem would be solved if multiple controllers were available, including one with fixed curves for acceleration / deceleration over current vehicle speed.

I'd appreciate any comment on how to fix my problem with 0.9.11. without the need to manually move the vehicles around. :)

Best regards
Robert

jackbart94, 2021-07-06:
We have answered to a [previous issue](https://github.com/carla-simulator/carla/issues/3839) from a user which had a similar problem. If this discussion isn't enough, let me know, however I think that there isn't much we can do.

brscholz, 2021-07-06:
Hi @jackbart94, of course I had searched the issues and found this thread. It feels like a different problem to me. I have no issues with reproducability if I do the exact same simulation under the same settings. It's just that when I run with different time steps, I will get different results and I guess that's mainly due to the controller in the traffic manager. It seems to behave very different although both time step settings are actually very low and should allow for a reliable control of the vehicle speed.

jackbart94, 2021-07-06:
The similarity I was talking about is that we suggest using a fixed_time_step around 0.05, or also 0.1. Setting it to 0.005 will probably mess with the physics engine, but @DSantosO can probably confirm this.

brscholz, 2021-07-06:
Thank you for the quick response, @jackbart94. Actually I would doubt that decreasing the time step blows up a physics engine that runs stable at larger time steps - that doesn't usually happen in physical modelling. However my time steps are not subject to changes anyway, I need my images exactly every 27 ms and my sensor values exactly every 5 ms. 

Seems I will have to go back to one simulation with cameras and sensors and stick with a time step of 1 ms if I want synchronous camera and sensor timings. :/ @DSantosO I would really appreciate your input, maybe we're still missing something.

brscholz, 2021-07-09:
I happen to have found a solution that fits my needs for now: increasing the substepping number to a larger value resulted in vehicle positions which only differ by a few centimeters instead of meters throughout the first 30 secs of my case.

settings.max_substeps=30

This keeps things comparable for 27 ms and 5 ms time step. :slight_smile: 
By the way, is there a reason why this substep number should range between 1 and 16?

closed 2021-11-18 by jackbart94