carla-simulator/carla #4375: Make launch error on Ubuntu18.04

seuwcs, 2021-07-07:
Hi, friends. I am installing carla on Ubuntu 18.04, carla version is 0.9.10, the UE4 version is 4.24
I have followed the instructions with ：https://carla.readthedocs.io/en/latest/build_linux/
and I have made PythonAPI sucessfully, but after that I run : make launch.
then the error occured.
The error is listed as the figure shows. I will appreciate it if someone can give me some help. Thanks a lot.

[image]

corkyw10, 2021-07-07:
Hi @seuwcs, you are referencing the wrong documentation. Our docs are versioned. If you are installing 0.9.10 please start from scratch using `make clean` and follow the instructions [here](https://carla.readthedocs.io/en/0.9.10/build_linux/)

seuwcs, 2021-07-22:
Thank you very much, the problem has been solved by your advice.

closed 2021-07-22 by corkyw10