carla-simulator/carla #4379: camera sensor HDR output

Domsall, 2021-07-08:
Over 3 years ago there was an issue regarding a 16-bit image: #157.

I couldn't find any updates on this topic. So my question is:
Has it since then become possible to output HDR images from the camera sensor and if not, is this a feature you are looking into in the near future?

It would be great to see this in the "Community friendly features".

Domsall, 2021-09-15:
I would be very thankful to get an update on this topic :)

Domsall, 2022-01-10:
what is the state of this topic?

roque2205, 2022-02-17:
I have made my own implementation of an hdr camera which delivers 16bit float pixel values instead of 8 bit int pixels. And then I found out that I only get 256 different float values. This topic was marked as stale, but maybe @Axel1092 or @DSantosO have any hints for me?

Domsall, 2022-02-17:
I am also still here :)
The reason is that the picture taken with the Scene-Capturer is transmitted from the server-side via 8 bit values.
I just do not know how much time this takes to implement or if an extension on this topic is coming soon.

Domsall, 2022-02-18:
You only changed the data format, the underlying data stayed the same (256 different values = 8 bit).

There may be 2 reasons:

- You still depend on the 8 bit Scene-Capturer from Carla
- You haven't changed the game settings to 12 or 16 bit. The standard game runs in 8 bit and the Scene Capturer just captures a photo of this game scene.

I would love to understand how/where you changed the files to get the 16 bit values.

Domsall, 2022-02-18:
Are you working with the Unreal Editor or directly changing the C++-Files?

From my understanding you have already done the difficult part -> Changing the Carla-server and -client files.

You could create a copy of the current 2D capturer and change some values in the Editor (similar to https://www.youtube.com/watch?v=uv0jYHLirbg). Or have you already done this?
Then the sensor probably needs to be put into the Sensor-Blueprint.

brscholz, 2022-02-18:
I'm doing everything in C++, as the original camera sensors themselves are created there, too.

Right now I am sure that, when the camera on the server side reads pixel values as float 16bit, I do retrieve these values using the Python API, without any loss from data type conversion.

I also changed the render target of my camera:

```
AHDRCamera::AHDRCamera(const FObjectInitializer &ObjectInitializer)
  : Super(ObjectInitializer)
{
  Enable16BitFormat(true);
  AddPostProcessingMaterial(
      TEXT("Material'/Carla/PostProcessingMaterials/PhysicLensDistortion.PhysicLensDistortion'"));
  CaptureRenderTarget->RenderTargetFormat = ETextureRenderTargetFormat::RTF_RGBA16f;
  CaptureRenderTarget->OverrideFormat = ETextureRenderTargetFormat::RTF_RGBA16f;
  CaptureRenderTarget->CompressionSettings = TextureCompressionSettings::TC_HalfFloat;
}
```
What the guy in the video does is pretty much the same as is done in _SceneCaptureSensor.cpp_. From my understanding, **ReadSurfaceFloatData** should now read more than 256 different colors from that render target... But it does not.

Do the materials in the level need special treatment?

brscholz, 2022-02-18:
I'll create a branch right after lunch and let you know then. :)

brscholz, 2022-02-18:
https://github.com/brscholz/carla/tree/brscholz_HDR_Camera

I guess there will be some compilation issues due to unity builds. I had to set it up from scratch because my local version includes a lot of stuff that I don't want to share. :) I'll try and compile it myself now and see what errors show up.

brscholz, 2022-02-21:
Hey @Domsall, are you able to access the code on my branch? The current version should compile, run and deliver images of data type uint16. Just use "sensor.camera.hdr" and retrieve by "np.frombuffer(data, dtype=np.uint16)". Looking forward to your feedback!

Domsall, 2022-02-22:
I just finished my implementation... I have a lot of work to do the next two days, so I will probably upload it on Friday.

Already a big thanks to you for giving this a push. There were only a few things missing/problematic. A few notes in advance:

1. Casting caused a lot of errors, because the Unreal-FFloat16 values operate differently than one with basic C++ knowledge would expect.
2. I would at least transfer 3xfloat16's from the server to the client (better floats, because more common) and not clip the values on the server's side to uint16_t. Depending on parameters, the floating point values can actually be 2-3x times higher than 1.0. If we transfer the full floating point values, the tone mapping and clipping can be done on the clients side according to the clients needs.
3. The option to render with "HDR" must be set (In the CaptureSensor-Code). It is currently set to LDR. The normal rgb-camera actually also benefits from this change (the tone mapping is then done my Unreal internally and the output is the normal 255-rgb but in a better light range). Maybe this could be made a changeable option for all cameras.

brscholz, 2022-02-23:
Thanks for your input!

I transferred floats initially, but I have made the experience that the additional payload in server-client communication from sending the images slows down the simulation noticeably. Back then I hadn't considered float values getting larger than 1 and I think that is a good reason for switching back to float now.

I also changed SDR to HDR, didn't notice that before.

Domsall, 2022-02-28:
I will upload the work tomorrow.
The only issue I am still having is that in 0.9.11 the resulting image (downscaled to 8bit in manual_control) is a lot brighter than in 0.9.13. The postprocessing seems to have changed between the versions. In 0.9.13, the image from manual_control (the client) is a lot darker than what you see on the server (CarlaUE4).
You will need to play around with the values and/or find that bug :)

Domsall, 2022-03-01:
Could you please grant me access to your repository?

Domsall, 2022-03-01:
I also added a few lines, which are currently "outcommented". You can use those lines for messages and debugging.

IbrahimYang, 2022-03-08:
Hi Domsall,
I'm also struggling with rendering HDR images in Carla. May you tell me your modification is working or not? If yes, may you share this core with us? Thanks！

Domsall, 2022-03-08:
Take a look into brscholz's branch, all the changes were made there.

IbrahimYang, 2022-03-16:
Thanks for your suggestion, we test your given code and it works well. We have some additional questions about the HDR camera
(1) I'm wondering about the dynamic range of the 16bit floating-point image generated by UE4. Could you please tell us what's the maximum value of UE4 outputs? 
(2) What does the FP16 image output by HDRcamera represent for, a higher dynamic range RGB or original texture rendered by UE4?

Domsall, 2022-03-16:
(1) Please take a look into the Unreal options: https://docs.unrealengine.com/4.27/en-US/RenderingAndGraphics/HDRDisplayOutput/
That really depends on your options.
(2) The Rendering happens in HDR anyway... what Unreal then does is the post-processing and tone mapping to bring it down to 8bit (Unreal calculates in floating point values). Even if an image is 16bit, what you are seeing depends on your graphics card and display. Textures can be 8bit png-images, so without some kind of weather/sun/environment you would only really get 8bit-range values back even with FP16. With PBR-materials and -textures you should get better results: https://docs.unrealengine.com/4.27/en-US/RenderingAndGraphics/Materials/PhysicallyBased/

So to conclude: You are getting a higher dynamic range RGB in all cases, but it depends on what your display can show, your materials and the post-processing options.

ztysdu, 2022-10-13:
Hi, I'm recently researching how to generate HDR images with carla, and I only found this [UE4 document](https://docs.unrealengine.com/4.26/en-US/RenderingAndGraphics/HDRDisplayOutput/) ,carla code
[image]
and your issue on the web. Can you tell me how you set up HDR mode? Thank you very much!

madhu1sept, 2022-12-05:
Hi, I  am also struggling with HDR camera. I need to capture image using float camera(e.g: R,G, B value range in between 26800 to 26832 for white image). Currently I am getting values it as 1(general range is coming between 0 and 1). So if I follow the above steps can I achieve the following range.In my case I dont want to do any manipulation in captured image(capture white image in this range) using HDR camera. 
Second question is why we used RTF_RGBA16f and not RTF_RGBA32f.
It will be really helpful if you can guide regarding this it is bit urgent.
Thanks in advance

Domsall, 2022-12-06:
we used RTF_RGBA16f to have a value of 64bit per pixel. The value is a floating point between 0 and 1, you can also change that value on the clients side to be between 0 and 65535 (uint16).

You will find all information you need to add a HDR camera here in this post.

madhu1sept, 2022-12-06:
Thanks for quick reply, but in  my case customer dont want to change value in Client side. He want to get the pixel value in higher range(may be as it is captured by float camera) when we extract from RenderTarget.
So I have two queries
1) Is there a way in carla or unreal side to disable Tone mapping and post processing. If it is possible can you please guide how to do it.
2) Is there any specific reason to limit the value between 0 and 1.
3) Why we are not using RTF_RGBA32f (I tried to modify code but it is causing crash)
Thanks for your help

Domsall, 2022-12-06:
1. If you have a customer, you are getting paid for your work. You can change many options, please read the posts above thoroughly. Information about tone mapping, etc. can be found here.
2. The value is normalized between 0 and 1 with steps of of 1/65535.
3. Again, 64bit... you may run into problems when you want to transfer 128bit pixels (with the buffer, serializing, deserializing, etc.)

I cannot help with these issues, you need to find out and fix them yourself.

madhu1sept, 2022-12-06:
Ok thanks for the info will look into the pointers you provided.

m-abdelkarim, 2023-02-16:
@Domsall  @brscholz Thanks for the great work. May I ask whether you are planning to open a pull request for that change? Many Carla Users would be happy about seeing that feature in the upcoming releases.

Domsall, 2023-02-16:
Sorry, not from my side or at least not in the near future. I am currently not working on this topic anymore and there are still some issues that need to be solved:
- Hard coded values need to be changeable (I needed to recompile everytime when switching between HDR/LDR)
- There seems to be an issue with the lighting when using HDR and the output is LDR
- Somebody would need to add some tone mapping / postprocessing features

See 

> (1) Please take a look into the Unreal options: https://docs.unrealengine.com/4.27/en-US/RenderingAndGraphics/HDRDisplayOutput/
That really depends on your options.
(2) The Rendering happens in HDR anyway... what Unreal then does is the post-processing and tone mapping to bring it down to 8bit (Unreal calculates in floating point values). Even if an image is 16bit, what you are seeing depends on your graphics card and display. Textures can be 8bit png-images, so without some kind of weather/sun/environment you would only really get 8bit-range values back even with FP16. With PBR-materials and -textures you should get better results: https://docs.unrealengine.com/4.27/en-US/RenderingAndGraphics/Materials/PhysicallyBased/
> So to conclude: You are getting a higher dynamic range RGB in all cases, but it depends on what your display can show, your materials and the post-processing options.

brscholz, 2023-02-16:
I agree with Dominik, there's still work to do to get it ready for a pull request and I personally lack the time. I got out of it what I needed for my project, but feel free to extend our work and create a merge request yourself.

m-abdelkarim, 2023-02-16:
Thanks for your input