carla-simulator/carla #4380: Question/Problem by importing a map

LucasFKobernic, 2021-07-08:
Hi everyone,

I'm trying to import a new map in CARLA. I followed the steps described here:

https://carla.readthedocs.io/en/0.9.11/tuto_A_add_map/

It works fine until the cration of bin file. I'm facing 2 problems:

1. When the Package is build, no directory called "Nav" is created. Should i create it manually or a error happend here?
2. The .bin file are not being generated. I run the command "build.bat map_file". It's shows me "Loading" and it ended without doing something. 

[image]

Can anyone help me?

Thanks

corkyw10, 2021-07-12:
Can you follow the [updated instructions](https://carla.readthedocs.io/en/latest/tuto_A_add_map/add_map_source/#generate-pedestrian-navigation) for generating the pedestrian nav and see if it helps? Some times these issues arise because the meshes were not exported correctly to the `.obj`file.

LucasFKobernic, 2021-07-12:
hello @corkyw10

Thanks for the answer.

I did the entire process from the beginning and then it worked. I noticed that when I have more than one map (.obj + .xodr) in the Dist directory, the "build.bat" sometimes doesn't work.

I have some questions about the map, which I've been facing in recent days:

1. Autopilot: so far I get it, autopilot is actually controlled by TrafficManager, right? And in this case, RoutePlanner doesn't play a role here. I'm trying to generate vehicles in autopilot mode, but 2 problems are occurring:
        1.1 My vehicle simply stands still and doesn't move. But if I run "spawn_npc.py" before and then run my code, it moves.
        1.2 On a map without traffic lights, vehicles are behaving strangely: they are going in circles or just in a straight line, not following the lane.

2. BehaviorAgent + RoutePlanner: I noticed that BehaviorAgent only works when I have a RoutePlanner. I don't know why, but when I insert my map some lanes don't have a RoutePlanner (I don't know if it's a problem on the RoadRunner side or in the insertion process). I would like to enter a RoutePlanner manually, is it possible? Is there any documentation?

Thanks for the help!

corkyw10, 2021-08-06:
Hey @LucasFKobernic sorry for the delay. 

1. You're right TM is autopilot. 
1.1. Can you show how you are setting your vehicle to autopilot?
1.2. Is that on your own custom map? Can you provide an example of what is happening?

2. We have some documentation about the Behavior and Basic agents [here](https://carla.readthedocs.io/en/0.9.12/adv_agents/). Let us know if it helps you or if you have any additional questions

closed 2021-11-18 by jackbart94