carla-simulator/carla #4382: No collision using a specific vehicle

pirib, 2021-07-09:
Hi,

As the title says - if you spawn the vehicle at index 1 e.g.:

`filter('vehicle')[1]`

then collision detection does not work. Tested my script with other vehicles, and it works as intended:

`
# Spawn and attach the collision sensor
col_bp = self.bp_lib.find('sensor.other.collision')
self.col_sensor = self.world.spawn_actor(col_bp, carla.Transform(), attach_to = self.actor)

# Start listening:
self.col_sensor.listen(lambda event: self.on_col_event(event) )

# Update the info about collision
def on_col_event(self, event):
        print( "Collission!")

`

gsmario, 2021-07-12:
Hello, thanks for the info! I've checked the 5 first vehicles of our list and all of them have their collision. 
We are going to look on more in depth about this problem and then let you know! 
Thanks!

qhaas, 2021-07-13:
Was [synchronous mode](https://carla.readthedocs.io/en/0.9.11/adv_synchrony_timestep) in use when this occurred?

pirib, 2021-07-14:
Oh yes, it was, following settings were applied:

`        settings = self.world.get_settings()
        settings.substepping = True
        settings.max_substep_delta_time = 0.01
        settings.max_substeps = 10
        settings.fixed_delta_seconds = settings.max_substep_delta_time * settings.max_substeps
        settings.synchronous_mode = True
        
        self.world.apply_settings(settings)
`

DSantosO, 2021-07-14:
Hello @pirib,

Thank you for reporting the issue.
Which version are you using? Can you post the id of the blueprint car that is presenting the problem?
A couple of months ago, we detected two cars that had this problem and we fix it. If you are using `dev` branch it should not happen for any car.

pirib, 2021-07-14:
Hi, 

Vehicle id=vehicle.mercedesccc.mercedesccc

I just tested it again with the settings outlined above, still no collision detection. I am running Carla 0.9.11, Python 3.7.9, on Win 10.

PS: sorry, my bad, I now realise I should have included more info in the original post.

DSantosO, 2021-07-14:
Hello @pirib,

The world settings should not affect the sensor. That was one of the cars that we fixed. If you use `dev` branch or our nightly builds it should be working fine. 
Thanks again for reporting. I close the issue.

closed 2021-07-14 by DSantosO

qhaas, 2021-07-14:
Interesting bug... I wasn't aware this was a problem... does it impact 0.9.8?  If so, I might have to manually exclude them for our HPC runs which randomly select a vehicle blueprint.

DSantosO, 2021-07-15:
Hello @qhaas,
I have checked version 0.9.8 and there are two cars presenting the bug: the mustang and the charger.
I have prepared a small script to check it so you can test it in that version or any other. 

[check_collision_sensor.zip](https://github.com/carla-simulator/carla/files/6822164/check_collision_sensor.zip)

qhaas, 2021-07-15:
Thanks, @DSantosO .  I remember seeing a randomly selected vehicle apparently collide with something and the sensor didn't register a collision; I optimistically assumed it was a near miss, but this explains it.