carla-simulator/carla #4391: LNK2038: mismatch detected for 'RuntimeLibrary' error make PythonAPI on Windows 10

AdelinaK, 2021-07-12:
Hi. I was trying to build Carla and I've got this error on make PythonAPI 

xerces-c_3.lib(DOMStringPool.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MT_StaticRelease' in libcarla.obj
xerces-c_3.lib(XPathSymbols.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MT_StaticRelease' in libcarla.obj
xerces-c_3.lib(XSTypeDefinition.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MT_StaticRelease' in libcarla.obj
xerces-c_3.lib(XMLGrammarDescription.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MT_StaticRelease' in libcarla.obj
xerces-c_3.lib(FieldValueMap.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MT_StaticRelease' in libcarla.obj
   Creating library build\temp.win-amd64-3.7\Release\source/libcarla\libcarla.cp37-win_amd64.lib and object build\temp.win-amd64-3.7\Release\source/libcarla\libcarla.cp37-win_amd64.exp
LINK : warning LNK4098: defaultlib 'MSVCRT' conflicts with use of other libs; use /NODEFAULTLIB:library
build\lib.win-amd64-3.7\carla\libcarla.cp37-win_amd64.pyd : fatal error LNK1319: 523 mismatches detected
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\VC\\Tools\\MSVC\\14.16.27023\\bin\\HostX86\\x64\\link.exe' failed with exit status 1319

Windows 10
Carla version 9.11

Any ideas why? Thanks!

gelinhe, 2021-07-14:
I have the same issue.
libboost_filesystem-vc141-mt-x64-1_72.lib(directory.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
libboost_filesystem-vc141-mt-x64-1_72.lib(path_traits.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
libboost_filesystem-vc141-mt-x64-1_72.lib(operations.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
libboost_filesystem-vc141-mt-x64-1_72.lib(path.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
libboost_filesystem-vc141-mt-x64-1_72.lib(exception.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
libboost_filesystem-vc141-mt-x64-1_72.lib(codecvt_error_category.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
libboost_filesystem-vc141-mt-x64-1_72.lib(windows_file_codecvt.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
libboost_python37-vc141-mt-x64-1_72.lib(errors.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
libboost_python37-vc141-mt-x64-1_72.lib(class.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
libboost_python37-vc141-mt-x64-1_72.lib(from_python.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
libboost_python37-vc141-mt-x64-1_72.lib(registry.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
libboost_python37-vc141-mt-x64-1_72.lib(builtin_converters.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
libboost_python37-vc141-mt-x64-1_72.lib(arg_to_python_base.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
libboost_python37-vc141-mt-x64-1_72.lib(list.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
libboost_python37-vc141-mt-x64-1_72.lib(tuple.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
libboost_python37-vc141-mt-x64-1_72.lib(module.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
libboost_python37-vc141-mt-x64-1_72.lib(dict.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
libboost_python37-vc141-mt-x64-1_72.lib(function.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
libboost_python37-vc141-mt-x64-1_72.lib(stl_iterator.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
libboost_python37-vc141-mt-x64-1_72.lib(iterator.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
libboost_python37-vc141-mt-x64-1_72.lib(life_support.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
libboost_python37-vc141-mt-x64-1_72.lib(inheritance.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
libboost_python37-vc141-mt-x64-1_72.lib(enum.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
libboost_python37-vc141-mt-x64-1_72.lib(object_protocol.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
libboost_python37-vc141-mt-x64-1_72.lib(object_operators.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
libboost_python37-vc141-mt-x64-1_72.lib(pickle_support.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
libboost_python37-vc141-mt-x64-1_72.lib(str.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
libboost_python37-vc141-mt-x64-1_72.lib(function_doc_signature.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
osm2odr.lib(OSM2ODR.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
osm2odr.lib(NIFrame.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
osm2odr.lib(NILoader.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
osm2odr.lib(NBFrame.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
osm2odr.lib(GeoConvHelper.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
osm2odr.lib(MsgHandler.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
osm2odr.lib(NBNetBuilder.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
[... cut ...]
xerces-c_3.lib(XPathSymbols.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
xerces-c_3.lib(XSTypeDefinition.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
xerces-c_3.lib(XMLGrammarDescription.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)
xerces-c_3.lib(FieldValueMap.obj) : error LNK2038: 检测到“RuntimeLibrary”的不匹配项: 值“MD_DynamicRelease”不匹配值“MT_StaticRelease”(libcarla.obj 中)

AdelinaK, 2021-07-14:
Well I'm going to answer myself. The problem in my case was with the Windows SDK. I had a newer version (10) and it conflicts with the 8.1 required to build the simulator. I uninstalled the newer one and the rebuild worked for me.

corkyw10, 2021-07-22:
@AdelinaK I'm glad you were able to resolve the issue, although it's strange because I think the newer SDK shouldn't interfere. 

@gelinhe can you let us know if this resolves the issue for you?

closed 2023-06-02 by stale[bot]