carla-simulator/carla #4395: Build Carla from source on a headless linux server

lalani1996, 2021-07-13:
Hello all,

I am using a remote headless Linux server and want to implement the RSS sensor on the server. For the implementation of the RSS sensor, I have to build Carla from the source in the server. Has anyone built Carla on a headless server? I tried X11 forwarding to launch Unreal Engine but it didn't work. Any suggestions would be appreciated.

Thanks and Regards

qhaas, 2021-07-13:
Cliff Notes:  If you need a GUI rendering on a remote server in GNU/Linux and use of it remotely via X11 forwarding, one way is to use OpenGL and setup VirtualGL.

Details:
You likely were unable to x-forward the UnrealEditor GUI (i.e. `make launch`) because it uses [vulkan by default now](https://carla.readthedocs.io/en/0.9.11/adv_rendering_options/#vulkan-vs-opengl) which [does not support x-forwarding](https://github.com/VirtualGL/virtualgl/issues/37).  Many SSH implementations only support older versions of OpenGL out-of-the-box and those versions have been removed from newer versions of UnrealEngine.  The way rendering works with X11 forwarding and GPU acceleration is not intuitive, [this write-up by VirtualGL explains it](https://virtualgl.org/About/Background).

We build/run CARLA on headless GNU/Linux systems, including HPCs.  You can use [`make package`](https://carla.readthedocs.io/en/0.9.11/build_linux/#make-carla) to build CARLA releases similar to those provided by the vendor on systems without a GUI.

You can save sensor data to files during CARLA simulation runs, so no need for X11 or a GUI.  You invoke CARLA server using [`SDL_VIDEODRIVER=offscreen ./CarlaUE4.sh -opengl`](https://github.com/carla-simulator/carla/issues/225) to get OpenGL rendering without a GUI (or even X11, so completely headless).  Vulkan rendering to flat files can be achieved with an [upgrade to UE 4.25+](https://github.com/carla-simulator/carla/issues/3214#issuecomment-769140155)

lalani1996, 2021-07-14:
Thanks a lot for the explanation.

corkyw10, 2021-07-22:
There is some draft documentation [here](https://github.com/carla-simulator/carla/blob/corkyw10/docker-docs/Docs/adv_rendering_options.md#setting-off-screen-mode-versions-prior-to-0912) explaining the different rendering options. Have a look if you have any input or would like to test it.

corkyw10, 2021-08-05:
Closing this issue, if you find anything to mention regarding the new documentation, please open a new issue. Thanks!

closed 2021-08-05 by corkyw10