carla-simulator/carla #4404: Unable to set custom spawn point while creating world

DokurOmkar, 2021-07-14:
I am using Carla Version 0.9.11 and trying to create a vehicle in automatic_control.py at the same location every time I run this script here (This code snippet is in def restart of World class).

`while self.player is None:
            if not self.map.get_spawn_points():
                print('There are no spawn points available in your map/town.')
                print('Please add some Vehicle Spawn Point to your UE4 scene.')
                sys.exit(1)
            spawn_points = self.map.get_spawn_points()
            #spawn_point = random.choice(spawn_points) if spawn_points else carla.Transform()
            current_location = carla.Location(x=-2.419356, y=204.005676, z=0.001191) #coordinates taken from normal run
            spawn_point = carla.Transform(current_location, carla.Rotation())
            print("Automatic Control Debug => 6.2.2 => self.player => ", self.player)
            self.player = self.world.try_spawn_actor(blueprint, spawn_point)` 

When I run the above code it does not come out of the while loop as the player is still none. 

Please advise what is the right way to set the player at a custom location.

Thank you!

MattRowe18, 2021-11-09:
@glopezdiest could you take a look at this thanks

glopezdiest, 2021-11-09:
It's just a matter of the height. With z=0, the vehicle always collides with the ground, so it will never spawn, no matter how many tried you do. By increasing the height to 0.5, the spawn should work just fine

closed 2021-11-09 by glopezdiest