carla-simulator/carla #4408: Carla 0.9.11 Pygame shows that serve has a very low FPS at around 3

Kingsman1995, 2021-07-15:


Kingsman1995, 2021-07-15:
Hi, I currently use the new version 0.9.11, here is the configuration of my laptop:
1. Ubuntu 18.04
2. Alienware m15 r4 (RTX 3080; intel core i9-10th gen)

It seems that my laptop has no problem with its configuration, but when I run the Carla simulator, it gets slow, the FPS of the server is around 3, and the client FPS is about 62. It runs slowly in the pygame when I start the manual_control.py. Does anyone know what's the reason causing this problem?

jackbart94, 2021-07-16:
Are you using the packaged version, or are you in the Unreal Engine editor?

KamazGraphics, 2021-08-09:
Possible solution: you need to open Unreal Engine Editor settings and find "Use less CPU in background" - uncheck the box and everything should be fine.

wangchengkang, 2021-09-15:
Possible Solution, find /PythonAPI/Utils/config.py.  run this script, e.g.: python3 config.py --fps=20.

closed 2021-11-09 by MattRowe18