carla-simulator/carla #4409: [Question] - Collision / Collision Box

LucasFKobernic, 2021-07-15:
Hi everyone,

I have some questions about collision inside CARLA. I looked for answers under the documentation and here, but some parts still unclear for me.

So far i have understood, the collision sensor detects a collision when the boxes of my actors are overlaping. Before version 0.9.11 the bounding box was used and since version 0.9.11 a collision box is used for the vehicle, right?

When I go to the static mesh of the vehicle in UE I can select under "collision" 2 types of mesh: simple collision and complex collision. Which one is used as collision box? Is the simple collision?(under Blueprint -> Custom Collision -> Static Mesh with 'sc' in the name is selected, is that for simple collision?).

Is it possible to set which collision box to be used though the PythonAPI ? or i need to change the Blueprint inside UE ? In the API I only found a class called "BoundingBox" und under collision sensor there is no attribute that make reference to the collision box.

For the pedestrian the bounding box still being used, right?

Thanks

DSantosO, 2021-08-04:
Hello @LucasFKobernic,

Inside UE, the two collision types simple and complex are used depending on how the project and some objects are configured. 
In Carla, for cars, we are not using boxes but simplified versions of the meshes. The custom collision that you mentioned is not used for physics but for raycast sensors like Lidar and Radar. For collisions, there is another 'mesh' that you can see in the physics asset, select the Vehicle Base in the SkeletonTree and the collider will be highlighted in orange.  You can easily change in here, just right-clicking in the Vehicle Base and select Add Shape (or copy collider from another mesh). Unfortunately, this can only be done inside the physics asset.
For the pedestrian, is no the box, but a combination of capsules, you can see them in the physics asset of the pedestrians.

LucasFKobernic, 2021-08-05:
Hi @DSantosO 

Thanks for your answer! it was very helpful :)

closed 2021-08-05 by LucasFKobernic

AOOOOOA, 2023-02-20:
> 

Hi @DSantosO , I have a few questions about the collision determine. In this issue, you said that the collision is determined by the mesh/bounding box, but the [measurement](https://carla.readthedocs.io/en/stable/measurements/) shows that the collision sensor returns the collision intensity.  I just wondering about the relationship between them, in other words, how to determine the collision happened in Carla? What is the criterion?