carla-simulator/carla #4414: Set Autopilot

anas-abulehia, 2021-07-18:
I want to enable the autopilot mode for a vehicle. When I run the python script, it does not move . 

'''
    client = carla.Client('localhost', 2000)
    traffic_manager = client.get_trafficmanager(8000)
    traffic_manager.set_global_distance_to_leading_vehicle(1.0)
    traffic_manager.set_hybrid_physics_mode(True)
    traffic_manager.set_synchronous_mode(True)
    client.set_timeout(5.0)
    
    world = client.get_world()
    blueprint_library = world.get_blueprint_library()
    bp = blueprint_library.filter('cybertruck')[0]
    print(bp)

    spawn_point = random.choice(world.get_map().get_spawn_points())
    print(spawn_point);
    vehicle = world.spawn_actor(bp, spawn_point)
    vehicle.set_autopilot(True, traffic_manager.get_port())
'''

jackbart94, 2021-07-19:
Make sure to check in depth how we do this in ``spawn_npc.py`` (or ``generate_traffic.py`` in dev) before opening an issue. It's also very well documented on the docs.

anas-abulehia, 2021-07-19:
I have implemented the same approach as in spawn_npc.py , I use windows, so there is no dev file in my Carla

qhaas, 2021-07-19:
> I use windows, so there is no dev file in my Carla

I believe he was referring to the [generate_traffic.py in the dev branch](https://github.com/carla-simulator/carla/blob/dev/PythonAPI/examples/generate_traffic.py)

LucasFKobernic, 2021-07-20:
Hi @anas-abulehia 

Your code seems right to me. The only difference to my is that i wrote "world.tick()" after setting the autopilot, to be sure the server recieves the information.

anas-abulehia, 2021-07-20:
I have added world.tick(), but same it stops
``
client = carla.Client('localhost', 2000)
    traffic_manager = client.get_trafficmanager(8000)
    traffic_manager.set_global_distance_to_leading_vehicle(1.0)
    traffic_manager.set_hybrid_physics_mode(True)
    traffic_manager.set_synchronous_mode(True)
    client.set_timeout(5.0)
    world = client.get_world()
    blueprint_library = world.get_blueprint_library()
    bp = blueprint_library.filter('cybertruck')[0]
    spawn_point = random.choice(world.get_map().get_spawn_points())
    print(spawn_point);
    vehicle = world.spawn_actor(bp, spawn_point)
    vehicle.set_autopilot(True, traffic_manager.get_port())
    world.tick();``

jackbart94, 2021-07-20:
> I have added world.tick(), but same it stops
> ` client = carla.Client('localhost', 2000) traffic_manager = client.get_trafficmanager(8000) traffic_manager.set_global_distance_to_leading_vehicle(1.0) [...]

The world.tick() goes in a ``while true:`` loop. Please, check the ``spawn_npc.py`` script, this is clearly shown there.

closed 2021-11-09 by MattRowe18