carla-simulator/carla #4420: RuntimeError: rpc::rpc_error during call in function version

navjotkaurdhoat, 2021-07-21:
Hello Community,

I have downloaded CARLA_0.9.10 from GitHub Repository and extracted it under my user on Ubuntu 18.04 server. The CARLA Simulator was working fine as both server and Python clients were able to establish connections. But recently, I started getting the following error while running Python scripts.

> Traceback (most recent call last):
>   File "manual_control.py", line 1137, in <module>
>     main()
>   File "manual_control.py", line 1129, in main
>     game_loop(args)
>   File "manual_control.py", line 1045, in game_loop
>     world = World(client.get_world(), hud, args)
> RuntimeError: rpc::rpc_error during call in function version

I checked all the past messages on Discord and GitHub, but nothing is working out to solve this problem. I also checked the Python version in the egg file and on my system.

EGG file: carla-0.9.10-py3.7-linux-x86_64.egg
Python version: Python 3.7.5
CARLA: 0.9.10
OS: Ubuntu 18.04
Rendering Option: TurboVNC

Can anyone please help me with this? Any help would be appreciated.

MattRowe18, 2021-11-09:
@bernatx could you take a look at this thanks

6Lackiu, 2022-04-22:
Just tested solution!
I start manual_control.py encountered the same problem. I found that the egg file was not added to the environment variable. At first, it was like this: `export pythonpath = "${carla_root} / Python API / Carla /:" ${scenario_runner_root} ":" ${leaderboard_root} "`
Changed to: ` export pythonpath = "${carla_root} / Python API / Carla /:" ${scenario_runner_root} ":" ${leaderboard_root} ":" ${carla_root} / Python API / Carla / dist / carla-0.9.10-py3.7-linux-x86_64. Egg ": ${pythonpath}`
Finally, you should start manual_control.py through `python3` instead of `python`! This perfectly solves the problem, at least for me.
My environment is Ubuntu 16.04, Carla version is 0.9.10.1. I started Carla from a remote server, so I added `-opengl`.
I hope everyone can solve the problem!

closed 2025-02-19 by Blyron