carla-simulator/carla #4438: Sky Overexposure in UE4

JaredPC, 2021-07-26:
Hi friends and developers!

I am running Carla 0.9.12 (dev branch) on Unreal Engine 4.26.1 built from source on linux and I am having issues with the sky being extremely overexposed in the UE4 editor window and when running the simulation. An example of the overexposure in the editor window on Town3 is shown below.

[image]

This light seems to be emanating from the entire sky and occurs on all default Carla maps and one of my own custom maps as well. Until recently I had Carla and UE4 running fine without this lighting issue. Initially I thought it was an issue with the BP_sky blueprint (Skylight or Directional Light), but adjusting all the parameters do not seem to affect the overexposure aspect, although they do have an effect on the lighting of the buildings. I tried reinstalling the Carla assets but the lighting remains the same, which makes me think that it is some UE4 setting that I accidentally changed.

I saw issue #3417 which seems to be related, but the answer that @Madecu posted did not fix the overexposure.

Does anyone have experience or advice for fixing this issue?

heethesh, 2021-08-15:
Similar to #4527, I see this issue in the packaged version of CARLA too. Any solutions?

JaredPC, 2021-08-22:
I updated my Nvidia driver to 470 like was mentioned in #4527 and the lighting returned to normal. Thank you for your comment @heethesh

closed 2021-09-01 by JaredPC

likith3611, 2022-08-11:
I tried with all nvidia drivers from 470 to 515 still facing the smae issue

crisdesivo, 2022-10-02:
> 

Same here, any solution?

bapukokare, 2023-08-04:
I have the same issue, but I think how to reduce the unreal engine setting for expected environment