carla-simulator/carla #4470: make import doens't working

LucasFKobernic, 2021-07-30:
Hi, 

I have recntly intalled CARLA 0.9.11 in a new computer. I'm trying to import a map that i have created in RoadRunner. 

Unfortunalley by "make import ARGS="--package <package_name> --no-carla-materials" the package in Content folder aren't being created and no .json file appears in the Import folder. And i'm getting no error in Terminal:

`make import ARGS="--package EuroNCAP_map --no-carla-materials"
-[Setup]: [Batch params]: --boost-toolset msvc-14.1
-[Setup]: Asynchronous jobs:  16
-[Setup]: Boost toolset:      msvc-14.1
-[Setup]: Install directory:  "C:\Users\z239023\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "C:\Users\Username\carla\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "C:\Users\Username\carla\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "C:\Users\z239023\carla\Build\" --zlib-install-dir "C:\Users\z239023\carla\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "C:\Users\Username\carla\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "C:\Users\z239023\carla\Build\"
    -[install_rpclib]: A rpclib installation already exists.
    -[install_rpclib]: Delete "C:\Users\Username\carla\Build\rpclib-install\" if you want to force a rebuild.
    -[install_rpclib]: Exiting...
-[Setup]: Installing Google Test...
    -[install_gtest]: [Batch params]: --build-dir "C:\Users\z239023\carla\Build\"
    -[install_gtest]: A Google Test installation already exists.
    -[install_gtest]: Delete "C:\Users\Username\carla\Build\gtest-install\" if you want to force a rebuild.
    -[install_gtest]: Exiting...
-[Setup]: Installing "Recast & Detour"...
    -[install_recast]: [Batch params]: --build-dir "C:\Users\z239023\carla\Build\"
    -[install_recast]: A "Recast & Detour" installation already exists.
    -[install_recast]: Delete "C:\Users\Username\carla\Build\recast-cdce4e-install\" if you want to force a rebuild.
    -[install_recast]: Exiting...
-[Setup]: Installing Boost...
    -[install_boost]: [Batch params]: --build-dir "C:\Users\z239023\carla\Build\" --toolset msvc-14.1 --version 1.72.0 -j 16
    -[install_boost]: A Boost installation already exists.
    -[install_boost]: Delete "C:\Users\Username\carla\Build\boost-1.72.0-install\" if you want to force a rebuild.
    -[install_boost]: Exiting...
        1 Datei(en) kopiert.
        1 Datei(en) kopiert.
-[Setup]: Installing Xercesc...
    -[install_xercesc]: [Batch params]: --build-dir "C:\Users\Username\carla\Build\"
    -[install_xercesc]: A xerces installation already exists.
    -[install_xercesc]: Delete "C:\Users\z239023\carla\Build\xerces-c-3.2.3-install\" if you want to force a rebuild.
    -[install_xercesc]: Exiting...
-[Setup]: Creating "CMakeLists.txt.in"...
-[Setup]:
   ###########
   # SUCCESS #
   ###########

   IMPORTANT!

   All the CARLA library dependences should be installed now.
   (You can remove all "*-src" folders in C:\Users\z239023\carla\Build\ directory)

   You only need the ASSET PACK with all the meshes and textures.

   This script provides the assets for CARLA 0.9.11:
   You can download the assets from here:

       http://carla-assets.s3.amazonaws.com/20201222_232b876.tar.gz

   Unzip it in the "C:\Users\Username\carla\Unreal\CarlaUE4\Content\Carla\" folder.
   If you want another version, search it in C:\Users\Username\carla\Util\ContentVersions.txt.

-[BuildLibCarla]: [Batch params]: --server
-- Selecting Windows SDK version 10.0.17763.0 to target Windows 10.0.19042.
-- Build debug:   ON
-- Build release: ON
-- Build test:    ON
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/z239023/carla/Build/libcarla-visualstudio
Microsoft (R)-Build-Engine, Version 15.9.21+g9802d43bc3 für .NET Framework
Copyright (C) Microsoft Corporation. Alle Rechte vorbehalten.

  carla_server.vcxproj -> C:\Users\Username\carla\Build\libcarla-visualstudio\LibCarla\cmake\server\Release\carla_server.lib
  carla_server_debug.vcxproj -> C:\Users\Username\carla\Build\libcarla-visualstudio\LibCarla\cmake\server\Release\carla_server_debug.lib
  -- Install configuration: "Release"
-[BuildLibCarla]: LibCarla server has been successfully installed in "C:\Users\Username\carla\Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies\"!`

Have anyone any idea what is the problem here?

Thanks

fengyongwei, 2021-08-16:
unluckily, i have the same problem, don't know hwo to deal with it

bernatx, 2021-10-06:
Hi, can you give more details about the structure of files you are trying to import, and how your .json file is defined. It seems that the import script is not finding any .json file with info about what to import.
We are fixing this script, so if you give more details we can add more fixes if needed.
Regards

closed 2023-06-02 by stale[bot]