carla-simulator/carla #4476: Question: Camera vertical field of view

FranekStark, 2021-07-31:
A Question about the field of view of cameras in Carla.
When setting the fov parameter it sets both the horizontal as well as the vertical fov. Is there any possibility to set them individually?

Madecu, 2021-08-06:
Hi @FranekStark,
We are using unreal's default cameras and the Field of View they allow to modify only changes horizontal fov 
[image]

The FOV is fixed, but as the viewport size changes, the aspect ratio is allowed to change, adjusting how much you see in the non-constrained axis. FOV can be expressed either as a fixed horizontal FOV or a fixed vertical FOV, and the engine currently uses fixed horizontal FOV, so the vertical FOV changes as the aspect ratio changes.

I found a video that might be helpful if you want to modify carla's unreal project. But for now, you can't set the vFOV individually from the API

https://www.youtube.com/watch?v=ZVEfspETs84&ab_channel=HarryPearce

FranekStark, 2021-08-09:
@Madecu thanks a lot for your explanations. As the fov and the ratio are related to each other it makes indeed a lot of sense to set the v-fov via the image ratio.

closed 2021-08-09 by FranekStark

roshambo919, 2022-10-29:
Does this then imply square pixels?