carla-simulator/carla #4479: In synchronous mode, the action obtained by the get_control function has a delay of two time steps than the action of the apply_control application.

Roxanneoux, 2021-08-02:
CARLA 0.9.5
CARLA 0.9.6
CARLA 0.9.9

In synchronous mode, the action obtained by the get_control function has a delay of two time steps than the action applied by apply_control. However, the speed obtained by the get_velocity function has a delay of one time step than the speed applied by the set_velocity function. That is, it applies a control object on the next tick.
Is it a problem with the CARLA simulator?

Roxanneoux, 2021-08-07:
@corkyw10
Hi. Is this a problem with my code or a problem with the Carla simulator?

DSantosO, 2021-08-13:
Hello @Roxanneoux,
This is a known issue that was fixed with the new synchronous pipeline in 0.9.11. You can check the details in the [release notes of this version.](http://carla.org/2020/12/22/release-0.9.11/)

closed 2021-08-13 by DSantosO