carla-simulator/carla #4482: Remote access Carla from Mac

Saquib764, 2021-08-03:
CARLA version: 0.9.11
Platform/OS: Ubuntu for simulator / Mac for PythonAPI

I am running Carla simulator in a remote ubuntu system. I want to connect to carla server using a client in mac. 

But I get the following error while importing carla in python environment-

```
(carla) Saquibs-MacBook-Pro:examples saquib$ python manual_control_carsim.py 

/Users/saquib/work/CARLA_0.9.11/PythonAPI/examples/manual_control_carsim.py
/Users/saquib/work/CARLA_0.9.11/PythonAPI/examples
Traceback (most recent call last):
  File "manual_control_carsim.py", line 85, in <module>
    import carla
  File "/Users/saquib/work/CARLA_0.9.11/PythonAPI/carla/dist/carla-0.9.11-py3.7-linux-x86_64.egg/carla/__init__.py", line 8, in <module>
  File "/Users/saquib/work/CARLA_0.9.11/PythonAPI/carla/dist/carla-0.9.11-py3.7-linux-x86_64.egg/carla/libcarla.py", line 9, in <module>
  File "/Users/saquib/work/CARLA_0.9.11/PythonAPI/carla/dist/carla-0.9.11-py3.7-linux-x86_64.egg/carla/libcarla.py", line 7, in __bootstrap__
ImportError: dlopen(/Users/saquib/Library/Caches/Python-Eggs/carla-0.9.11-py3.7-linux-x86_64.egg-tmp/carla/libcarla.cpython-37m-x86_64-linux-gnu.so, 2): no suitable image found.  Did find:
        /Users/saquib/Library/Caches/Python-Eggs/carla-0.9.11-py3.7-linux-x86_64.egg-tmp/carla/libcarla.cpython-37m-x86_64-linux-gnu.so: unknown file type, first eight bytes: 0x7F 0x45 0x4C 0x46 0x02 0x01 0x01 0x00
        /Users/saquib/Library/Caches/Python-Eggs/carla-0.9.11-py3.7-linux-x86_64.egg-tmp/carla/libcarla.cpython-37m-x86_64-linux-gnu.so: unknown file type, first eight bytes: 0x7F 0x45 0x4C 0x46 0x02 0x01 0x01 0x00
```

Kin-Zhang, 2021-08-04:
If you using the mac how to install carla?
details at here: https://carla.readthedocs.io/en/latest/build_faq/#importerror-no-module-named-carla

Kin-Zhang, 2021-08-04:
And I also see here for 0.9.12 if you want to use mac for PythonAPI
https://carla.readthedocs.io/en/latest/start_quickstart/#carla-0912