carla-simulator/carla #4487: Center of mass location

lorenzo-medici7, 2021-08-04:
I'm trying to get the position of the vehicle's center of mass.

x and y values give a realistic result if applied to the center of the bounding box (e.g. y value is close to 0, x value varies depending on engine placement) but the z value is always negative and too low to be applied at the same location because it results in hilariously low CoG's for some vehicles.
Applying it to the z coordinate of the vehicle.get_location() value doesn't make any sense either because it looks like it is the level of the road and not the level of the geometric center or the center of mass.

Does anybody know the actual reference point for the center_of_mass location?

suresh-guttikonda, 2022-04-27:
Is there any update on this issue ? The COM value returned is different than the UE's vehicle blueprint.

sshawn9, 2025-01-09:
same question