carla-simulator/carla #4498: error while loading shared libraries: libomp.so.5

afterimagex, 2021-08-06:


mawi42, 2021-08-06:
Hi,

when I installed libomp5 it worked for me:

apt-get install libomp5

br 
mawi

guygo12345, 2021-08-08:
Got same problem when running on machine with no root permissions (so I can't install libomp5).
What can I do?

dmllr, 2021-08-11:
Same issue when running official 0.9.12 container. There's no issue with 0.9.11.

Installation of `libomp5` **inside** docker container works as a workaround, but apparently it should be fixed officially.

qhaas, 2021-08-13:
> Got same problem when running on machine with no root permissions (so I can't install libomp5).
> What can I do?
> 
> **EDIT**
> Tried to copy the .so file from another place and add it to the PATH, now I don't get the error anymore but the server is not running and the process exits

Deployed the `libomp5` package inside the container and it crashes with no meaningful error.

dmllr, 2021-08-13:
> Deployed the `libomp5` package inside the container and it crashes with no meaningful error.

It might be related to the graphics driver in Ubuntu. Still not able to run the container over Vulcan backend.

What is your system environment?

qhaas, 2021-08-13:
Fixed the above crash by `unset SDL_VIDEODRIVER`, I was running [xorg-less / headless](https://carla.readthedocs.io/en/0.9.12/adv_rendering_options/#setting-off-screen-mode-version-0912) with `./CarlaUE4.sh -vulkan -RenderOffscreen`.  I suspect others might be using the GUI.

Created P.R. #4538 to add the missing dependency.

guygo12345, 2021-08-15:
Thank you, the "-RenderOffscreen" option fixed my problem (I didn't notice the change)

qhaas, 2021-08-25:
> Thank you, the "-RenderOffscreen" option fixed my problem (I didn't notice the change)

More CLI of interest to those doing headless / xorg-less vulkan rendering... you can [select which GPU to render on](https://github.com/carla-simulator/carla/issues/3214#issuecomment-769140155) by using e.g. `-graphicsadapter=1` (for GPU index 1)... also, if you start getting [ALSA related warnings](https://github.com/carla-simulator/carla/issues/2820#issuecomment-888819712), use `-nosound`.

qhaas, 2021-09-09:
Looks like the P.R. was accepted into the dev branch, any chance we can get the 0.9.12 image rebuilt and pushed to dockerhub such that it will work out of the box?

yasser-h-khalil, 2021-09-17:
Hi,

I have carla 0.9.12 installed on ubuntu.
When executing the following line: ./CarlaUE4.sh
I get the following error: CARLA_0.9.12/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping: error while loading shared libraries: libomp.so.5: cannot open shared object file: No such file or directory

Carla was installed using "pip3 install carla"

Any help is appreciated.
Thanks

yasser-h-khalil, 2021-09-17:
> Hi,
>
> I have carla 0.9.12 installed on ubuntu.
> When executing the following line: ./CarlaUE4.sh
> I get the following error: CARLA_0.9.12/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping: error while [...]

Solved my issue by executing the following two lines:

sudo apt-get update
sudo apt-get install libomp5

hh0rva1h, 2021-11-30:
@bernatx It is fixed in the docker images, but not in the 0.9.12 and 0.9.13 release builds people fetch from Github. Could you fix those as well?

farhanhubble, 2022-01-01:
I faced this issue with 0.9.13 on Fedora 35. I had to install libomp-devel and then soft-link `/lib64/libomp.so` as `/lib54/libomp.so.5`.

Bernhard1999, 2022-11-09:
@bernatx Could you give an update whether the issue with the libomp.so.5 library has been fixed or will be fixed for the versions 0.9.12 and 0.9.13 release builds which people fetch from GitHub ?

kingDanonymous, 2024-07-08:
> Hi,
> 
> when I installed libomp5 it worked for me:
> 
> apt-get install libomp5
> 
> br mawi

Worked for me Thank you