carla-simulator/carla #4514: unable to parse the OpenDRIVE XML string

seuwcs, 2021-08-09:
Hello every one!

I have built carla 0.9.12 from source on Ubuntu 18.04 with the edition of Unreal Engine is 4.26.

I made maps in Roadrunner and exported to Unreal, got the 00.fbx and 00.xodr files, then import them to carla, when the map imported, it worked in UE4 like figure 1.

[image]

but when I run the script manual_control.py in the terminal, something went wrong, the error is :

ERROR: unable to parse the OpenDRIVE XML string 
RuntimeError: failed to generate map
  The server could not send the OpenDRIVE (.xodr) file:
  Make sure it exists, has the same name of your town, and is correct.
[image]

However,  I do place the file 00.xodr in the folder with the path is  carla/Unreal/CarlaUE4/Content/Carla/Maps/OpenDrive/ 
[image]

Besides,  I have tried to find solutions in 
https://github.com/carla-simulator/carla/issues/2257
but it did not work for me, I got puzzled.
who can give me some suggestions? any help is appreciated !

More importantly. I have another computer, Ubuntu18.04, carla version is 0.9.10, Unreal Engine 4.24, surprisingly, it worked.

I really don't know how could it can't work with carla 0.9.12 and  UE 4.26.

Any suggestions is appreciated! thank you very much!
@krmartin @sunpochin @Jazzinghen @cesarsouza @ItsTimmy @glopezdiest  @corkyw10 @XGodina

LucasFKobernic, 2021-08-10:
I'm having the same issues

seuwcs, 2021-08-12:
> I'm having the same issues

if you have solved the problem?

LucasFKobernic, 2021-08-12:
Not yet, 

I made the map insertion in CARLA 0.9.11 and then i copied the map folder in CARLA 0.9.12 Content folder and now i'm able to open and use the map. It is not the best solution, but it works for now.

seuwcs, 2021-08-12:
> Not yet,
> 
> I made the map insertion in CARLA 0.9.11 and then i copied the map folder in CARLA 0.9.12 Content folder and now i'm able to open and use the map. It is not the best solution, but it works for now.

It seems strange.

seuwcs, 2021-08-13:
> Not yet,
> 
> I made the map insertion in CARLA 0.9.11 and then i copied the map folder in CARLA 0.9.12 Content folder and now i'm able to open and use the map. It is not the best solution, but it works for now.

I have a question, the figure below describes the file that had been imported to UE4, in the folder, there do not exist the file that named exp0_BuiltData.uasset.

[image]

However, in the folder /carla/Unreal/CarlaUE4/Content/Carla/Maps. the map file built by carla default, do have the file whose suffix is "_BuiltData.uasset",

XGodina, 2021-08-16:
Hi @seuwcs, @LucasFKobernic !! When you import a RoadRunner map into CARLA, a folder of that map is created. Inside that folder, you should put the OpenDrive with their according folder.
[image]

seuwcs, 2021-08-17:
> Hi @seuwcs, @LucasFKobernic !! When you import a RoadRunner map into CARLA, a folder of that map is created. Inside that folder, you should put the OpenDrive with their according folder.
> [image]

Thanks for your advice. I have made a try, but I still got the same problem.

Firstly, I imported a map to carla from roadrunner in the folder Maps(/home/wcs/carla/Unreal/CarlaUE4/Content/Carla/Maps), like the following figure:

[image]

the new folder is named as "e1", under e1, the OpenDrive folder is created and the e1.xodr file was put under the OpenDrive folder, like this:

[image]

but when I clicked "Play" in UE4, and run the manual_control.py script under the path : carla/PythonAPI/examples, that question still exist, like this:

[image]

As you know, after the map was imported to carla, two files will be created, and usually the two files are named as : filename.umap and filename_BuiltData.uasset respectively, and they are correspond to each other. like this:

[image]

However,  in the folder ExportedMaps(/home/wcs/carla/Unreal/CarlaUE4/Content/Carla/ExportedMaps), there only exist one file named :e1.umap, basicially, the file e1.BuiltData.uasset should also be created.But it didn't.

[image]

What troubles me most is why the e1.BuiltData.uasset file is not created.

Besides, the e1.xodr file had been put in the folder OpenDrive in the path: /home/wcs/carla/Unreal/CarlaUE4/Content/Carla/Maps/OpenDrive  as well.

[image]

but the problem still exist (unable to parse the OpenDRIVE XML string). I really don't know why this happens.

if you know how this happened?Thanks for your advice.

XGodina, 2021-08-17:
I will talk with our engineers about that. Now almost all our team are on holiday. I'll let you know when I know something.

supavit-siriwan, 2021-09-28:
I got the same problem. And I fix by the following method and it woke fine.

I guess that CARLA lib is search ".xodr" from the location of ".umap".
So, I create "OpenDrive" folded beside the ".umap" file. And contain the ".xodr" file.
In my case is at "~/carla/CarlaUE4/Content/Carla/ExportedMaps"

----------

And for the compiled package. I do the same after complied.
In my case is at "~/carla/Dist/CARLA_Shipping_0.9.12-dirty/LinuxNoEditor/CarlaUE4/Content/Carla/ExportedMaps"
I think the compile script did not include this folder by default.

seuwcs, 2021-09-29:
Thank you for sharing your solution,  I will have a try. Very much thanks.

Phd-Ma, 2021-11-22:
I got the same problem.

werewolfdev, 2022-01-26:
Hi @supavit-siriwan @seuwcs  iam also trying to export map from road runner and use in CARLA build from source. AS of now I used export (carla) option , put them in the import folder and did make import. Now inorder to open the Unreal engine editor with the custom map could you please tell me what should be done from my end

jl940213, 2022-02-23:
I succeeded in solving the problem , you should create a folder named OpenDrive in /<carla_root>/Unreal/CarlaUE4/Content/Carla/Maps/test. The 'test' is your own roadrunner map package which contain many .uasset file .Then put the same named .xodr file into OpenDrive file. Then run the manual_control.py!

PPTaniya, 2022-08-25:
I am also facing the same issue. Tried the above provided solution. But it seems its not working for me. Please let me know any other solution is there. 

CARLA Version: 0.9.13
OS: ubuntu 18.04.06
Unreal Engine: 4.26

lpxiangyan9, 2022-08-26:
i have the same question,
i use TruevisionDesigner export .xodr file  and import to UE4. 
but running py code throw this error
CARLA Version: 0.9.13
OS:WIN 10
UE:UECARLA4

sangeetsu, 2022-09-20:
I have the exact same problem
Used RoadRunner to Import map. Used the same RoadRunner export on a different PC with the same versions of everything and it works there.
Ubuntu - 18 LTS
Carla 0.9.13
UE 4.26

Tried editing permissions (gave chmod 777) for the OpenDrive folders. It works if I run the server from the UE Editor. If I package it and try to use manual_control.py it throws this error:

> ERROR: unable to parse the OpenDRIVE XML string 
Traceback (most recent call last):
  File "manual_control_steeringwheel.py", line 871, in <module>
    main()
  File "manual_control_steeringwheel.py", line 863, in main
    game_loop(args)
  File "manual_control_steeringwheel.py", line 792, in game_loop
    world = World(client.get_world(), hud, args.filter)
  File "manual_control_steeringwheel.py", line 140, in __init__
    self.restart()
  File "manual_control_steeringwheel.py", line 162, in restart
    spawn_points = self.world.get_map().get_spawn_points()
RuntimeError: failed to generate map

supavit-siriwan, 2023-03-23:
@sangeetsu For packaging. You need to add path of custom OpenDrive folder to UE4 project config.

Config file: ~/carla/Unreal/CarlaUE4/Config/DefaultGame.ini

add
```
+DirectoriesToAlwaysStageAsUFS=(Path="RHT_Straight_10km/Maps/RHT_Straight_10km/OpenDrive")
```

under
```
+DirectoriesToAlwaysStageAsUFS=(Path="Carla/Config")
```

Note: RHT_Straight_10km is my custom map.

tmhempen, 2023-07-20:
I had the same problem (ERROR: unable to parse the OpenDRIVE XML string when running generate_traffic.py) and was able to resolve it by adding my custom map path (/Game/<my_map_package>/Maps/<my_map>/<my_map>) to MapPaths.txt (C:\carla\Unreal\CarlaUE4\Content\MapPaths.txt).

CARLA Version: 0.9.13
OS: WIN 10
UE: 4.26

Edelweissor, 2023-10-18:
You may find the solution here, [https://github.com/carla-simulator/carla/issues/4010#issuecomment-810349551](url).
The issue was happening due to lack of write permission for the server on the directory `CarlaUE4/Content/Carla/Maps/OpenDrive/` because the server needs to write a file there during parsing.

Danlaw-Nikhil, 2024-01-18:
Has anyone solved this issue , I have followed the procedures in the same way 
I have downloaded an osm file from https://www.openstreetmap.org/
created a .fbx file for the same using blender 
imported both of these files into unreal engine editor (Carla fork) [ But was unable to import the .xodr file]
now i am able to launch the CARLA , but again the map loads in CARLA , but whenever i try accessing the same through the Python APIs , world.get_map() returns an error stating : 
                 ERROR: unable to parse the OpenDRIVE XML string
                 RuntimeError: failed to generate map
 i am using 
 Ubuntu 20.04
 Unreal engine 4.26
 CARLA 0.9.15
 
 any suggestions are welcome.
 Thank you
[image]
[image]
[image]

josyulavt, 2024-02-09:
> @sangeetsu For packaging. You need to add path of custom OpenDrive folder to UE4 project config.
>
> Config file: ~/carla/Unreal/CarlaUE4/Config/DefaultGame.ini
>
> add
>
> ``` [...]

Hi thanks for the answer, I don't see any occurrences of `DirectoriesToAlwaysStageAsUFS` in my `DefaultEngine.ini ` , these are contents of my file 

```
[/Script/Engine.Engine]
bSmoothFrameRate=false
SmoothedFrameRateRange=(LowerBound=(Type="ERangeBoundTypes::Inclusive",Value=22),UpperBound=(Type="ERangeBoundTypes::Exclusive",Value=120))

[/Script/HardwareTargeting.HardwareTargetingSettings]
TargetedHardwareClass=Desktop
AppliedTargetedHardwareClass=Desktop
DefaultGraphicsPerformance=Maximum
AppliedDefaultGraphicsPerformance=Maximum

[/Script/EngineSettings.GameMapsSettings]
EditorStartupMap=/Game/Carla/Maps/Town10HD_Opt.Town10HD_Opt
GameDefaultMap=/Game/map_package/Maps/smartroad/smartroad.smartroad
ServerDefaultMap=/Game/Carla/Maps/Town10HD_Opt.Town10HD_Opt
GlobalDefaultGameMode=/Game/Carla/Blueprints/Game/CarlaGameMode.CarlaGameMode_C
GameInstanceClass=/Script/Carla.CarlaGameInstance
TransitionMap=/Game/Carla/Maps/Town10HD_Opt.Town10HD_Opt
GlobalDefaultServerGameMode=/Game/Carla/Blueprints/Game/CarlaGameMode.CarlaGameMode_C

[/Script/Engine.RendererSettings]
r.DefaultFeature.MotionBlur=True
r.BasePassOutputsVelocity=True
r.BasePassForceOutputsVelocity=False
r.AllowStaticLighting=True
r.DiscardUnusedQuality=True
r.DefaultFeature.Bloom=False
r.DefaultFeature.AmbientOcclusion=False
r.DefaultFeature.AmbientOcclusionStaticFraction=False
r.DefaultFeature.AutoExposure=False
r.CustomDepth=3
r.Streaming.PoolSize=4000
r.TextureStreaming=True
r.GenerateMeshDistanceFields=True
r.DistanceFieldBuild.EightBit=False
r.DistanceFieldBuild.Compress=False
r.DistanceFields.AtlasSizeXY=1024
r.DistanceFields.AtlasSizeZ=2048
r.DefaultFeature.AutoExposure.ExtendDefaultLuminanceRange=True
r.DefaultFeature.AntiAliasing=2
r.VirtualTextures=True

[/Script/AIModule.AISense_Sight]
bAutoRegisterAllPawnsAsSources=False
bAutoRegisterNewPawnsAsSources=False

[/Script/NavigationSystem.RecastNavMesh]
RuntimeGeneration=Static

[/Script/AIModule.CrowdManager]
MaxAgents=1000

[/Script/LinuxTargetPlatform.LinuxTargetSettings]
SpatializationPlugin=
ReverbPlugin=
OcclusionPlugin=
-TargetedRHIs=SF_VULKAN_SM5
-TargetedRHIs=GLSL_430
+TargetedRHIs=SF_VULKAN_SM5
+TargetedRHIs=GLSL_430
[... 61 lines cut ...]
CompressionOverrides=(bOverrideCompressionTimes=False,DurationThreshold=5.000000,MaxNumRandomBranches=0,SoundCueQualityIndex=0)
bUseAudioStreamCaching=False
CacheSizeKB=0
MaxChunkSizeOverrideKB=0
bResampleForDevice=False
MaxSampleRate=0.000000
HighSampleRate=0.000000
MedSampleRate=0.000000
LowSampleRate=0.000000
MinSampleRate=0.000000
CompressionQualityModifier=0.000000
AutoStreamingThreshold=0.000000
SoundCueCookQualityIndex=-1

[/Script/Engine.CollisionProfile]
+Profiles=(Name="CustomSensorCollision",CollisionEnabled=QueryOnly,bCanModify=True,ObjectTypeName="SensorObject",CustomResponses=((Channel="WorldStatic",Response=ECR_Ignore),(Channel="WorldDynamic",Response=ECR_Ignore),(Channel="Pawn",Response=ECR_Ignore),(Channel="Visibility",Response=ECR_Ignore),(Channel="Camera",Response=ECR_Ignore),(Channel="PhysicsBody",Response=ECR_Ignore),(Channel="Vehicle",Response=ECR_Ignore),(Channel="Destructible",Response=ECR_Ignore),(Channel="SensorObject"),(Channel="SensorTrace")),HelpMessage="Used for custom collision meshes for objects that has very complex meshes but we want them to appear in raycast based sensors")
+DefaultChannelResponses=(Channel=ECC_GameTraceChannel1,DefaultResponse=ECR_Ignore,bTraceType=False,bStaticObject=False,Name="SensorObject")
+DefaultChannelResponses=(Channel=ECC_GameTraceChannel2,DefaultResponse=ECR_Ignore,bTraceType=True,bStaticObject=False,Name="SensorTrace")
+DefaultChannelResponses=(Channel=ECC_GameTraceChannel3,DefaultResponse=ECR_Overlap,bTraceType=True,bStaticObject=False,Name="OverlapChannel")
+EditProfiles=(Name="BlockAll",CustomResponses=((Channel="SensorObject"),(Channel="SensorTrace")))
+EditProfiles=(Name="OverlapAll",CustomResponses=((Channel="SensorObject",Response=ECR_Overlap),(Channel="SensorTrace",Response=ECR_Overlap)))

```
`smartroad` is my custom map 
I have placed the map in `Import` folder and compiled carla as specified on the documentation, All the folders have rwx permissions, I'm unsure what's happening, Did you succeed in solving this? can you please help? thank you

sangeetsu, 2024-02-09:
@josyulavt could you explain in detail the steps you followed to make the custom map and a list of files you have in your Import folder?

sangeetsu, 2024-02-09:
> Has anyone solved this issue , I have followed the procedures in the same way I have downloaded an osm file from https://www.openstreetmap.org/ created a .fbx file for the same using blender imported both of these files into unreal engine editor (Carla fork) [ But was unable to import the .xodr file] now i am able to launch the CARLA , but again the map loads in CARLA , but whenever i try accessing the same through the Python APIs , world.get_map() returns an error stating : ERROR: unable to parse the OpenDRIVE XML string RuntimeError: failed to generate map i am using Ubuntu 20.04 Unreal engine 4.26 CARLA 0.9.15
> 
> any suggestions are welcome. Thank you ![1](https://private-user-images.githubusercontent.com/154869281/297631120-5ad79f6f-b101-4552-960e-b9f804b17d8c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc1MTA2ODYsIm5iZiI6MTcwNzUxMDM4NiwicGF0aCI6Ii8xNTQ4NjkyODEvMjk3NjMxMTIwLTVhZDc5ZjZmLWIxMDEtNDU1Mi05NjBlLWI5ZjgwNGIxN2Q4Yy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIwOVQyMDI2MjZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iMjA1NjQyY2JiZGQzMTFiOTQ2ZTEyNWFlMDQxYjI4NTlmMzlhZDJiYjQwMzE4MDAyZWNkYzQxMTJiZDFjYjNlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.EuQGjBRgMsfWjYZ-sVYVeIKUfqDPb_-Jdf2aO00Ui3w) ![2](https://private-user-images.githubusercontent.com/154869281/297631281-46e05499-7c22-4191-bb37-c66ce57888e8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc1MTA2ODYsIm5iZiI6MTcwNzUxMDM4NiwicGF0aCI6Ii8xNTQ4NjkyODEvMjk3NjMxMjgxLTQ2ZTA1NDk5LTdjMjItNDE5MS1iYjM3LWM2NmNlNTc4ODhlOC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIwOVQyMDI2MjZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xOThiMjcyN2EzYjA0ZmUyN2E2YzBkMDJmZmExNzYwZjQ1MTZkOGE2NDZjM2RmNzlkYWQ5NGNkNjY4MzQ2MjUyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.RGw_zlymrr4rMuDZ5lMrYSm-x2JuB3gaDwDtDT4HB2s) ![3](https://private-user-images.githubusercontent.com/154869281/297631590-f8d5e802-8991-4022-9b60-3fd07c293570.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc1MTA2ODYsIm5iZiI6MTcwNzUxMDM4NiwicGF0aCI6Ii8xNTQ4NjkyODEvMjk3NjMxNTkwLWY4ZDVlODAyLTg5OTEtNDAyMi05YjYwLTNmZDA3YzI5MzU3MC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIwOVQyMDI2MjZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yZTY5MTQ3ODg2ZmY0YzI0ODYyMDI1OWU1MjBjNWY4YzQ1N2E0NDg1ZDBlNGQ4ZjVhNGE3MDU2YzkyMTk0M2NkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.ZdG_rR40fYPOyO86Ca-5xjGbc8PIV6nfwnLId9ruxOQ)

I had pasted my xodr(OpenDRIVE) file from the RoadRunner export into my packaged version sample33\LinuxNoEditor\CarlaUE4\Content\Carla\Maps\OpenDrive\

josyulavt, 2024-02-09:
> @josyulavt could you explain in detail the steps you followed to make the custom map and a list of files you have in your Import folder?

1. Export osm pbf from your favorite website. 
2. Import osm into roadrunner and generate .fbx, .xodr and other files under export as Carla Map 
3. Place the files into "Import" folder in carla 
4. Do make import 
5. Do make launch 
6. You can now see your map in file -> open levels -> yourmapname
7. make launch 
8. make package

sangeetsu, 2024-02-09:
> > @josyulavt could you explain in detail the steps you followed to make the custom map and a list of files you have in your Import folder?
> 
> 1. Export osm pbf from your favorite website.
> 2. Import osm into roadrunner and generate .fbx, .xodr and other files under export as Carla Map
> 3. Place the files into "Import" folder in carla
> 4. Do make import
> 5. Do make launch
> 6. You can now see your map in file -> open levels -> yourmapname
> 7. make launch
> 8. make package
>    I had to manually paste the smartroad.xodr file in Maps/OpenDrive, but I still get the same error

Did you add your custom map into the list of packaged maps in UE Editor? I remember setting my custom map as default on there.

josyulavt, 2024-02-09:
> Did you add your custom map into the list of packaged maps in UE Editor? I remember setting my custom map as default on there.

I followed this tutorial https://www.youtube.com/watch?v=ctRlzUM8QdM and yes I set it as default too..

sangeetsu, 2024-02-09:
I see them doing that at 3:07, not sure what else could cause the problem

sangeetsu, 2024-02-09:
> > I see that happening at 3:07, not sure what else could have caused the problem.
> 
> I have not added any lights or buildings to the map, could that be an issue?

That shouldn't be an issue. It should load default lighting if you haven't added it.

josyulavt, 2024-02-09:
> > > I see that happening at 3:07, not sure what else could have caused the problem.
> > 
> > 
> > I have not added any lights or buildings to the map, could that be an issue?
> 
> That shouldn't be an issue. It should load default lighting if you haven't added it.

I see.. thanks @sangeetsu

josyulavt, 2024-02-13:
I resolved this by placing the contents generated in the` Content/<map_name>/OpenDrive/<map_name>.xodr` into `Dist/.../CarlaUE4/.../Content/<map_name>/OpenDrive/<map_name>.xml`

sergiopaniego, 2024-09-09:
> @sangeetsu For packaging. You need to add path of custom OpenDrive folder to UE4 project config.
>
> Config file: ~/carla/Unreal/CarlaUE4/Config/DefaultGame.ini
>
> add
>
> ``` [...]

This solution worked for me! :rocket:

meilaoliu, 2024-10-14:
> > > > I see that happening at 3:07, not sure what else could have caused the problem.我在 3：07 看到这种情况发生，不确定还有什么可能导致问题。
> > > 
> > > 
> > > I have not added any lights or buildings to the map, could that be an issue?我没有在地图上添加任何灯光或建筑物，这会是个问题吗？
> > 
> > 
> > That shouldn't be an issue. It should load default lighting if you haven't added it.这应该不是问题。如果您尚未添加默认照明，它应该会加载它。
> 
> I see.. thanks  明白了。。谢谢@sangeetsu

> I resolved this by placing the contents generated in the` Content/<map_name>/OpenDrive/<map_name>.xodr` into `Dist/.../CarlaUE4/.../Content/<map_name>/OpenDrive/<map_name>.xml`我通过将 ` Content/<map_name>/OpenDrive/<map_name>.xodr` `Dist/.../CarlaUE4/.../Content/<map_name>/OpenDrive/<map_name>.xml`

OMG,Thank you so much,That sloved my problem perfectly!