carla-simulator/carla #4516: Can I use C++ API after package installation?

jiwung, 2021-08-09:
I have installed Carla via package installation: https://carla.readthedocs.io/en/latest/start_quickstart/
I am wondering if I can implement C++ client program, in this setup, such as https://github.com/carla-simulator/carla/tree/master/Examples/CppClient. The package installation does NOT seem to install C++ API libraries such as `libcarla_client.a`, `librpc.a` ... I couldn't find them on my machine.
Do I have to do source build instead?

dmllr, 2021-08-16:
The package you've mentioned is for Carla Server only. In order to make C++ Client API libraries you have to do source build yourself. Or, as an option, make Carla a dependency of your app using CMake for instance.

Really appreciate if Carla team would supply client libraries as a standalone package too.

jiwung, 2021-08-16:
Hi @dmllr. 
Thanks for your answer.
I strongly agree with you that it would be really appreciated if Carla team supplies client libraries as a standalone package too.

closed 2023-06-02 by stale[bot]