carla-simulator/carla #4522: Problem to import map in CARLA 0.9.12

LucasFKobernic, 2021-08-10:
Hi everyone,

I just installed CARLA 0.9.12 in my computer and i'm trying to insert a map that i created in RoadRunner.

I'm following the steps described here:

https://carla.readthedocs.io/en/0.9.12/tuto_M_add_map_source/

but by running "make import" no folder is being created in the Content folder. And the output in terminal is not the same as before in 0.9.11. So i think something is wrong:
```

make import  ARGS="--no-carla-materials"
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --no-carla-materials
-[Setup]: Asynchronous jobs:  16
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Install directory:  "D:\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "D:\carla\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "D:\carla\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "D:\carla\Build\" --zlib-install-dir "D:\carla\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "D:\carla\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "D:\carla\Build\"
    -[install_rpclib]: A rpclib installation already exists.
    -[install_rpclib]: Delete "D:\carla\Build\rpclib-install\" if you want to force a rebuild.
    -[install_rpclib]: Exiting...
-[Setup]: Installing Google Test...
    -[install_gtest]: [Batch params]: --build-dir "D:\carla\Build\"
    -[install_gtest]: A Google Test installation already exists.
    -[install_gtest]: Delete "D:\carla\Build\gtest-install\" if you want to force a rebuild.
    -[install_gtest]: Exiting...
-[Setup]: Installing "Recast & Detour"...
    -[install_recast]: [Batch params]: --build-dir "D:\carla\Build\"
    -[install_recast]: A "Recast & Detour" installation already exists.
    -[install_recast]: Delete "D:\carla\Build\recast-0b13b0-install\" if you want to force a rebuild.
    -[install_recast]: Exiting...
-[Setup]: Installing Boost...
    -[install_boost]: [Batch params]: --build-dir "D:\carla\Build\" --toolset msvc-14.2 --version 1.72.0 -j 16
    -[install_boost]: A Boost installation already exists.
    -[install_boost]: Delete "D:\carla\Build\boost-1.72.0-install\" if you want to force a rebuild.
    -[install_boost]: Exiting...
        1 Datei(en) kopiert.
        1 Datei(en) kopiert.
-[Setup]: Installing Xercesc...
    -[install_xercesc]: [Batch params]: --build-dir "D:\carla\Build\"
    -[install_xercesc]: A xerces installation already exists.
    -[install_xercesc]: Delete "D:\carla\Build\xerces-c-3.2.3-install\" if you want to force a rebuild.
    -[install_xercesc]: Exiting...
        1 Datei(en) kopiert.
-[Setup]: Installing Sqlite3
    -[install_sqlite3]: [Batch params]: --build-dir "D:\carla\Build\"
    -[install_sqlite3]: A Sqlite3 installation already exists.
    -[install_sqlite3]: Delete "D:\carla\Build\sqlite3-install" if you want to force a rebuild.
    -[install_sqlite3]: Exiting...
        1 Datei(en) kopiert.
-[Setup]: Installing PROJ
    -[install_proj]: [Batch params]: --build-dir "D:\carla\Build\"
    -[install_proj]: A PROJ installation already exists.
    -[install_proj]: Delete "D:\carla\Build\proj-install" if you want to force a rebuild.
    -[install_proj]: Exiting...
        1 Datei(en) kopiert.
-[Setup]: Creating "CMakeLists.txt.in"...
-[Setup]:
   ###########
   # SUCCESS #
   ###########

   IMPORTANT!

   All the CARLA library dependences should be installed now.
   (You can remove all "*-src" folders in D:\carla\Build\ directory)

   You only need the ASSET PACK with all the meshes and textures.

   This script provides the assets for CARLA 0.9.12:
   You can download the assets from here:

       http://carla-assets.s3.amazonaws.com/20210730_564bcdc.tar.gz

   Unzip it in the "D:\carla\Unreal\CarlaUE4\Content\Carla\" folder.
   If you want another version, search it in D:\carla\Util\ContentVersions.txt.

-[BuildLibCarla]: [Batch params]: --server
-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19042.
-- Build debug:   ON
-- Build release: ON
-- Build test:    ON
-- Configuring done
-- Generating done
-- Build files have been written to: D:/carla/Build/libcarla-visualstudio
Microsoft (R)-Build-Engine, Version 16.10.2+857e5a733 für .NET Framework
Copyright (C) Microsoft Corporation. Alle Rechte vorbehalten.

  carla_server.vcxproj -> D:\carla\Build\libcarla-visualstudio\LibCarla\cmake\server\Release\carla_server.lib
  carla_server_debug.vcxproj -> D:\carla\Build\libcarla-visualstudio\LibCarla\cmake\server\Release\carla_server_debug.lib
  -- Install configuration: "Release"
-[BuildLibCarla]: LibCarla server has been successfully installed in "D:\carla\Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies\"!

```
have anyone any idea what is going wrong?

todorovic1, 2021-08-10:
You can import the map manually from Content Browser in CarlaUE4.

LucasFKobernic, 2021-08-10:
I have tried it too. But then i'm getting this error:

`ERROR: unable to parse the OpenDRIVE XML string`

seuwcs, 2021-08-11:
> You can import the map manually from Content Browser in CarlaUE4.

it didn't work.

kpatil84, 2021-08-11:
You can import map with RoadRunner plugin and save map in Content/Carla/maps.

seuwcs, 2021-08-12:
> You can import map with RoadRunner plugin and save map in Content/Carla/maps.

but this error may occur:
ERROR: unable to parse the OpenDRIVE XML string

closed 2022-05-02 by stale[bot]