carla-simulator/carla #4525: Error LNK2001 at make PythonAPI when building OSM2ODR

roque2205, 2021-08-11:
Hi everyone,

I fail to run "make PythonAPI". There are a lot of xercesc errors ocurring when the compiler is trying to build osm2odr.

```
osm2odr.vcxproj -> E:\TEST_Carla_912\Build\osm2odr-visualstudio\src\Release\osm2odr.lib
  Building Custom Rule E:/TEST_Carla_912/Build/om2odr-source/CMakeLists.txt
  -- Install configuration: "Release"
  -- Installing: E:/TEST_Carla_912/PythonAPI/carla/dependencies/lib/osm2odr.lib
  -- Installing: E:/TEST_Carla_912/PythonAPI/carla/dependencies/include/OSM2ODR.h
-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "E:\TEST_Carla_912\PythonAPI\carla\dependencies\"
-[BuildPythonAPI]: [Batch params]: --py3
Building Python API for Python 3.
compiling:
  - source/libcarla/libcarla.cpp
running bdist_egg
running egg_info
creating source\carla.egg-info
writing source\carla.egg-info\PKG-INFO
writing dependency_links to source\carla.egg-info\dependency_links.txt
writing top-level names to source\carla.egg-info\top_level.txt
writing manifest file 'source\carla.egg-info\SOURCES.txt'
reading manifest file 'source\carla.egg-info\SOURCES.txt'
writing manifest file 'source\carla.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
running install_lib
running build_py
creating build
creating build\lib.win-amd64-3.7
creating build\lib.win-amd64-3.7\carla
copying source\carla\__init__.py -> build\lib.win-amd64-3.7\carla
copying source\carla\command.py -> build\lib.win-amd64-3.7\carla
running build_ext
building 'carla.libcarla' extension
creating build\temp.win-amd64-3.7
creating build\temp.win-amd64-3.7\Release
creating build\temp.win-amd64-3.7\Release\source
creating build\temp.win-amd64-3.7\Release\source\libcarla
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Idependencies/include -IC:\Users\Robert\AppData\Local\Programs\Python\Python37\include -IC:\Users\Robert\AppData\Local\Programs\Python\Python37\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt" /EHsc /Tpsource/libcarla/libcarla.cpp /Fobuild\temp.win-amd64-3.7\Release\source/libcarla/libcarla.obj /experimental:external /external:W0 /external:I dependencies/include/system /DBOOST_ALL_NO_LIB /DBOOST_PYTHON_STATIC_LIB /DBOOST_ERROR_CODE_HEADER_ONLY /D_WIN32_WINNT=0x0600 /DHAVE_SNPRINTF /DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true /MD
libcarla.cpp
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\link.exe /nologo /INCREMENTAL:NO /LTCG /DLL /MANIFEST:EMBED,ID=2 /MANIFESTUAC:NO /LIBPATH:dependencies/lib /LIBPATH:C:\Users\Robert\AppData\Local\Programs\Python\Python37\libs /LIBPATH:C:\Users\Robert\AppData\Local\Programs\Python\Python37\PCbuild\amd64 "/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\lib\x64" "/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\lib\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\lib\um\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.19041.0\ucrt\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.19041.0\um\x64" "/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\lib\x64" "/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\lib\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\lib\um\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.19041.0\ucrt\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.19041.0\um\x64" /EXPORT:PyInit_libcarla build\temp.win-amd64-3.7\Release\source/libcarla/libcarla.obj /OUT:build\lib.win-amd64-3.7\carla\libcarla.cp37-win_amd64.pyd /IMPLIB:build\temp.win-amd64-3.7\Release\source/libcarla\libcarla.cp37-win_amd64.lib shlwapi.lib Advapi32.lib ole32.lib shell32.lib E:\TEST_Carla_912\PythonAPI\carla\dependencies/lib\Detour.lib E:\TEST_Carla_912\PythonAPI\carla\dependencies/lib\DetourCrowd.lib E:\TEST_Carla_912\PythonAPI\carla\dependencies/lib\Recast.lib E:\TEST_Carla_912\PythonAPI\carla\dependencies/lib\rpc.lib E:\TEST_Carla_912\PythonAPI\carla\dependencies/lib\libboost_filesystem-vc142-mt-x64-1_72.lib E:\TEST_Carla_912\PythonAPI\carla\dependencies/lib\libboost_python37-vc142-mt-x64-1_72.lib E:\TEST_Carla_912\PythonAPI\carla\dependencies/lib\zlib.lib E:\TEST_Carla_912\PythonAPI\carla\dependencies/lib\libpng.lib E:\TEST_Carla_912\PythonAPI\carla\dependencies/lib\carla_client.lib E:\TEST_Carla_912\PythonAPI\carla\dependencies/lib\osm2odr.lib
   Bibliothek "build\temp.win-amd64-3.7\Release\source/libcarla\libcarla.cp37-win_amd64.lib" und Objekt "build\temp.win-amd64-3.7\Release\source/libcarla\libcarla.cp37-win_amd64.exp" werden erstellt.
osm2odr.lib(GeoConvHelper.obj) : error LNK2001: Unresolved external symbol "proj_create".
osm2odr.lib(GeoConvHelper.obj) : error LNK2001: Unresolved external symbol "proj_destroy".
osm2odr.lib(GeoConvHelper.obj) : error LNK2001: Unresolved external symbol "proj_trans".
osm2odr.lib(GeoConvHelper.obj) : error LNK2001: Unresolved external symbol "proj_torad".
osm2odr.lib(GeoConvHelper.obj) : error LNK2001: Unresolved external symbol "proj_todeg".
osm2odr.lib(OptionsIO.obj) : error LNK2001: Unresolved external symbol ""public: static void * __cdecl xercesc_3_2::XMemory::operator new(unsigned __int64)" (??2XMemory@xercesc_3_2@@SAPEAX_K@Z)".
osm2odr.lib(XMLSubSys.obj) : error LNK2001: Unresolved external symbol ""public: static void * __cdecl xercesc_3_2::XMemory::operator new(unsigned __int64)" (??2XMemory@xercesc_3_2@@SAPEAX_K@Z)".
osm2odr.lib(SUMOSAXReader.obj) : error LNK2001: Unresolved external symbol ""public: static void * __cdecl xercesc_3_2::XMemory::operator new(unsigned __int64)" (??2XMemory@xercesc_3_2@@SAPEAX_K@Z)".
osm2odr.lib(SUMOSAXReader.obj) : error LNK2001: Unresolved external symbol ""public: static void __cdecl xercesc_3_2::XMemory::operator delete(void *)" (??3XMemory@xercesc_3_2@@SAXPEAX@Z)".
osm2odr.lib(OptionsIO.obj) : error LNK2001: Unresolved external symbol ""public: static void __cdecl xercesc_3_2::XMemory::operator delete(void *)" (??3XMemory@xercesc_3_2@@SAXPEAX@Z)".
osm2odr.lib(XMLSubSys.obj) : error LNK2001: Unresolved external symbol ""public: static void __cdecl xercesc_3_2::XMemory::operator delete(void *)" (??3XMemory@xercesc_3_2@@SAXPEAX@Z)".
osm2odr.lib(OptionsLoader.obj) : error LNK2001: Unresolved external symbol ""public: static void __cdecl xercesc_3_2::XMemory::operator delete(void *)" (??3XMemory@xercesc_3_2@@SAXPEAX@Z)".
osm2odr.lib(GenericSAXHandler.obj) : error LNK2001: Unresolved external symbol ""public: static void __cdecl xercesc_3_2::XMemory::operator delete(void *)" (??3XMemory@xercesc_3_2@@SAXPEAX@Z)".
osm2odr.lib(OptionsIO.obj) : error LNK2001: Unresolved external symbol ""public: __cdecl xercesc_3_2::SAXParser::SAXParser(class xercesc_3_2::XMLValidator * const,class xercesc_3_2::MemoryManager * const,class xercesc_3_2::XMLGrammarPool * const)" (??0SAXParser@xercesc_3_2@@QEAA@QEAVXMLValidator@1@QEAVMemoryManager@1@QEAVXMLGrammarPool@1@@Z)".
[... cut ...]
osm2odr.lib(SUMOSAXAttributesImpl_Xerces.obj) : error LNK2001: Unresolved external symbol ""public: static void __cdecl xercesc_3_2::XMLString::release(char16_t * *,class xercesc_3_2::MemoryManager * const)" (?release@XMLString@xercesc_3_2@@SAXPEAPEA_SQEAVMemoryManager@2@@Z)".
build\lib.win-amd64-3.7\carla\libcarla.cp37-win_amd64.pyd : fatal error LNK1120: 52 unresolved externals

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "E:\TEST_Carla_912\PythonAPI\carla\dist"!
```

Does anyone have a hint at why this is happening? I only have one Python version, VS2019 is installed, I followed all the steps in the Windows build manual.

Best regards
Robert

LucasFKobernic, 2021-08-11:
Hi, 

I faced the same problem. I don't now why this is happening but to counter that I simply copied the osm2odr folder that I had in CARLA 0.9.11 into the  Build folder from version 0.9.12 and ran "make PythonAPI" again and it worked.

roque2205, 2021-08-11:
Hi, I'll give this a try, thanks. :)

brscholz, 2021-08-12:
This worked for me, seems to be a general bug with 9.12 osm2odr source.

TrinhTuanHung2021, 2021-08-13:
> Hi,
> 
> I faced the same problem. I don't now why this is happening but to counter that I simply copied the osm2odr folder that I had in CARLA 0.9.11 into the Build folder from version 0.9.12 and ran "make PythonAPI" again and it worked.

Could you explain in detail the process of executing "make PythonAPI" command? I use Win 10 and type that command on cmd, it gives me an error of not executing. Thanks

LucasFKobernic, 2021-08-13:
Hi @TrinhTuanHung2021 

You need to run the command "make PythonAPI" in the x64 Native Tools Command Prompt for VS 2019 for CARLA 0.9.12, not on cmd. 

You need to be inside of CARLA root directory too. The process is really good described at the documentation: https://carla.readthedocs.io/en/0.9.12/build_windows/

let me now, if you got it :)

TrinhTuanHung2021, 2021-08-13:
> Hi @TrinhTuanHung2021
>
> You need to run the command "make PythonAPI" in the x64 Native Tools Command Prompt for VS 2019 for CARLA 0.9.12, not on cmd.
>
> You need to be inside of CARLA root directory [...]

Thank you for replying to me. But when i run in the x64 Native Tools Command Prompt in the root directory folder, it has an error like this

[image]

Now when I run Carla it has an error
[image]
Thanks

LucasFKobernic, 2021-08-16:
have you installed make ? (version 3.81)

roque2205, 2021-09-02:
For the osm2odr LNK2001 error I found that "make clean", delete "Build" folder, "make PythonAPI" results in an error - but then when I run "make PythonAPI" again, it works. Don't change anything manually, just run "make PythonAPI" again. This works 100% - at least for for my system.

Prabesh-lingdao, 2024-01-24:
> Hi,
> 
> I faced the same problem. I don't now why this is happening but to counter that I simply copied the osm2odr folder that I had in CARLA 0.9.11 into the Build folder from version 0.9.12 and ran "make PythonAPI" again and it worked.

Could you please send me the osm2odr folder. I am also having the same issue.

TroyZhangFawang, 2024-08-21:
has anyone solved this problem?