carla-simulator/carla #4526: Python Packaging - Suggestions

stonier, 2021-08-11:
**CARLA version**: 0.9.12
**Platform/OS**: Ubuntu 18.04/20.04 but applies to any
**Problem you have experienced**: The python experience doesn't work out of the box (esp. with scenario runner)

**What you expected to happen**:

`pip install carla` into a python venv -> examples, agent modules and scenario runner works. In addition, other client-side workflows have access to `agents` and `examples`.

**What actually happened**:

* The carla wheel / pip package doesn't include the agent libraries, nor the examples.
* To get scenario runner to work involves hand-curating a bunch of python paths to multiple points (eggs, agents folder, examples folder, carla folder)
* Ended up hand-crafting our own python package/deb for client/scenario workflows.

**What could be done?**:

There are three components desirable to have client side - 1) carla, 2) agents and 3) examples. The current wheel / pip package only includes 1). You could consolidate all of these into the carla package/wheel itself, or spin these into separate packages/wheels.

Doing this would enable a client-side installation to work with `scenario_runner` and be very useful to non-scenario_runner workflows as well.

qhaas, 2021-08-13:
[This has come up before](https://github.com/carla-simulator/carla/issues/2077#issuecomment-651989689), I agree.

I end up just [patching the setup.py file](https://github.com/carla-simulator/carla/files/6984628/add_agents_util.txt) to include them, but could use some namespace reworking (e.g. `carla.agents`, `carla.util`, etc).

stonier, 2021-08-13:
> but could use some namespace reworking (e.g. carla.agents, carla.util, etc).

Agreed. `agents` & co. are name collisions waiting to happen for `scenario_runner` and other users.

MattRowe18, 2021-12-01:
@stonier thank you for your suggestions.

The Agents package is currently in progress and should be released soon, possibly with 0.9.14 or maybe before. It will be named "carla-agents" for pip installation although it will probably be named "agents" for import. There will only be a clash if you happen to be using a package imported as "agents". I'm not aware of one at the moment, the "agents" package for tensorflow is actually imported as "tf_agents" even though it is identified by pip as "agents"

closed 2021-12-01 by MattRowe18