carla-simulator/carla #4527: strange light artifact in version 0.9.12

wendytwh, 2021-08-11:
Hi everyone, 

I downloaded and installed the CARLA 0.9.12 version from the repository. When I launch the simulator running the 
`./CarlaUE4.sh`, the render view has a strong light artifact making the scene looks overexpose. 
When the weather is configured into cloudy night, it also has gives strange lightning effect. Does someone face same issue or have any idea what might be the reason?

os Ubuntu18.04 
graphics: Intel Xe Graphics (TGL GT2)

[image]
[image]

heethesh, 2021-08-13:
same issue here

XGodina, 2021-08-17:
@wendytwh I think this is causing Vulkan. Have you tried OpenGL ? ./CarlaUE4.sh -opengl

heethesh, 2021-08-18:
@XGodina seems like `-opengl` is not longer available on 0.9.12 right? I actually tried running the same exact build (with Vulkan) on another system with the more recent Nvidia driver 470 and the RGB camera looked fine but still see the over-exposure issue on system driver version 440 or 460.

XGodina, 2021-08-19:
@heethesh Yes, you're right. From what you tell me I see that it is a drivers problem.

chenhengwei1999, 2022-04-28:
I have the same the same issue. After I start the ./CarlaUE4.sh, there are some warnings in the <terminal>. The contents is as below. 

*****<Warning Start>*****
"4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

../src/intel/isl/isl.c:2105: FINISHME: ../src/intel/isl/isl.c:isl_surf_supports_ccs: CCS for 3D textures is disabled, but a workaround is available."
*****<Warning End>*****

I guess the problem is associating the Intel devices?( I am not familar with this).

My env is as below:
Ubuntu 20.04
NVIDIA-drivers 470.103.01 CUDA-version 11.4  NVIDIA-SMI 470.103.01

Looking For Help.

likith3611, 2022-08-11:
Hii even I am facing the same issue with the latest version of carla can you guys please help me with this?

crisdesivo, 2022-10-02:
Same issue here, any possible workaround?

chengyuew, 2022-10-10:
same issue here.

heethesh, 2022-10-11:
did you'll try upgrading with nvidia driver versions? or a source build?

chengyuew, 2022-10-11:
i solve this problem by installing 0.9.11

lucalazzaroni, 2022-12-12:
Same issue here.

redspry, 2023-02-07:
same problem...

lucalazzaroni, 2023-02-09:
Managed to solve by adding `export VK_ICD_FILENAMES="/usr/share/vulkan/icd.d/nvidia_icd.json"` to my `.bashrc`

ItsXianqi, 2023-03-24:
> 

great thanks for your advice! problem fixed.

ItsXianqi, 2023-03-24:
> Managed to solve by adding `export VK_ICD_FILENAMES="/usr/share/vulkan/icd.d/nvidia_icd.json"` to my `.bashrc`

great thanks for your advice! problem fixed.

redspry, 2023-06-02:
I was actually able to resolve this by running with -preferNvidia I think there is a vulkan driver issue?
`./CarlaUE4.sh  -preferNvidia`

-preferNvidia: This is an optional flag. If your system has multiple GPUs (for example, an integrated Intel GPU and a discrete Nvidia GPU), this flag tells Unreal Engine to use the Nvidia GPU, which is generally more powerful and better suited to handling CARLA's graphics requirements.

bainisuoci, 2024-07-12:
Is there any suitable solution?

Blyron, 2024-11-08:
Try the solutions above 
./CarlaUE4.sh  -preferNvidia
and 
export VK_ICD_FILENAMES="/usr/share/vulkan/icd.d/nvidia_icd.json" to my .bashrc

closed 2025-08-26 by Blyron