carla-simulator/carla #4529: warning: version mismatch detected. You are trying to connect to a simulator that might be incompatible with this api

TrinhTuanHung2021, 2021-08-12:
Hi. I use win 10, I just downloaded Carla to use. when I ran Python in the example section, I got this error. I want to ask how to fix it, thanks 

Thanks for contributing to CARLA!

If you are reporting an issue, please use the following outline:

CARLA version: 0.9.11
Platform/OS: Window
The problem you have experienced:warning: version mismatch detected. You are trying to connect to a simulator that might be incompatible with this api
What you expected to happen: how to fix it
Steps to reproduce:
Other information (documentation you consulted, workarounds you tried):

If you are asking a question please make sure your question was not asked before
by searching among the existing issues and checking the CARLA forum https://github.com/carla-simulator/carla/discussions. Also, make sure you have read our documentation and FAQ at http://carla.readthedocs.io. 

If your question is about creating content (assets, levels) you'll most likely
have all the info you need in the Unreal Engine's documentation
https://docs.unrealengine.com

Please, keep the threads focused and single-themed. Make them easy to find
for everyone.

-->
[image]

TrinhTuanHung2021, 2021-08-12:
I searched in the forum and found this error on Linux and fixed it with the command "make pythonAPI". But in win 10 when I do this command cmd gives an error.

qhaas, 2021-08-13:
Looks like you are trying to use carla client 0.9.12 with carla server 0.9.12; you either need to downgrade your pythonapi client version to 0.9.11 or upgrade your server binaries to 0.9.12.

TrinhTuanHung2021, 2021-08-14:
> Looks like you are trying to use carla client 0.9.12 with carla server 0.9.11; you either need to downgrade your pythonapi client version to 0.9.11 or upgrade your server binaries to 0.9.12.

I followed the instructions to run the command" make PythonAPI" but it didn't work.

qhaas, 2021-08-15:
Not familiar with the build procedure on Windows, but if `make` is not on the `PATH`, something is likely amiss in your [build environment](https://carla.readthedocs.io/en/0.9.11/build_windows/#visual-studio-2017).  Did you ` In the Workloads section, choose Desktop development with C++`?

Also, once you build the whl file, you will then need to `pip install` it into your python environment, that make target does not automatically deploy the client whl to your python environment.

yh300, 2021-11-05:
> Not familiar with the build procedure on Windows, but if `make` is not on the `PATH`, something is likely amiss in your [build [...]

How can i download this pythonapi client version (such as 0.9.11),thank you for your reply.

MattRowe18, 2021-12-01:
In or #

> How can i download this pythonapi client version (such as 0.9.11),thank you for your reply.

@yh300 hi

You should use pip to install the wheel that you have built in the build process "pip3 install /carlaRoot/PythonAPI/carla/dist/carla-#.#.#-#####.whl".

Alternatively you can specify the version to install via PyPi as such "pip3 install carla==0.9.12" although you should ensure that this matches your build version. In general, we would advise using pip if you are installing CARLA from packages, however, if you are building Carla from source, we would advise to also build the PythonAPI and use the wheel that is built instead of pip, this will ensure version consistency) 

If you want to check your CARLA PythonAPI version, use "pip3 show carla"

closed 2021-12-01 by MattRowe18