carla-simulator/carla #4532: Problem with DVS sensor

stephenjia, 2021-08-12:
Hi,

I run into a problem when using simulated DVS sensor. There is too much space between lines when the car is not driving slowly.

When it runs less than 30km/hr, it looks like this
[image]

[image]

When it runs at more than 100km/hr, it looks like this
[image]

[image]

I wonder whether this is normal? Is it caused due to the limitation of rendering speed?  How can I avoid rendering such kind of event data?

Thanks a lot!

DSantosO, 2021-08-13:
Hello @stephenjia,

The DVS sensor compares the rendered image in the current frame with the one rendered in the previous one. When the car is going too fast the two images change a lot so the result that you see is the expected one. If you are using a variable time step you cannot control this, but if you are using a fixed delta time you can decrease it so you render images more often.

stephenjia, 2021-08-14:
Hi @DSantosO ,

Thank you for your reply. So does it mean if we would like to get more continuous events, we should use fixed delta time and reduce it to a small number?

Thanks!

DSantosO, 2021-08-31:
Yes, exactly.

closed 2021-08-31 by DSantosO

Shahid1Malik, 2024-08-05:
Hi @stephenjia. Can you help me in simulating DVS in carla?

Shahid1Malik, 2024-08-05:
@DSantosO  does existing DVS EC simulator in Carla solely rely on RGB images to generate events, or it also uses other data like depth