carla-simulator/carla #4533: Get information of vehicles

hailuu684, 2021-08-12:
hello everyone,
I want to get information of vehicles like: velocity, acceleration, location,... but what I see in the spawn_npc.py, it uses the command to spawn the vehicles, and as I read the docs, command.Spawn_actor does not have method to retrieve the information. 

```
 vehicle_batch = []
        for n, transform in enumerate(spawn_points):
            if n >= args.number_of_vehicles:
                break
            blueprint = random.choice(blueprints)
            if blueprint.has_attribute('color'):
                color = random.choice(blueprint.get_attribute('color').recommended_values)
                blueprint.set_attribute('color', color)
            if blueprint.has_attribute('driver_id'):
                driver_id = random.choice(blueprint.get_attribute('driver_id').recommended_values)
                blueprint.set_attribute('driver_id', driver_id)
            blueprint.set_attribute('role_name', 'autopilot')

            # prepare the light state of the cars to spawn
            light_state = vls.NONE
            if args.car_lights_on:
                light_state = vls.Position | vls.LowBeam | vls.LowBeam

            # spawn the cars and set their autopilot and light state all together
            vehicle_batch.append(SpawnActor(blueprint, transform)
                .then(SetAutopilot(FutureActor, True, traffic_manager.get_port()))
                .then(SetVehicleLightState(FutureActor, light_state)))
```

Instead, I spawn the ego vehicle by using world.try_spawn_actor, then I can easily get the information using get_location(),....
However, I use for loop to spawn vehicles as is in the tutorial.py, the vehicles do not move at all. I do not understand why because they are also set autopilot?
    ```
    for _ in range(0, 10):
            transform.location.x += 8.0

            bp = random.choice(blueprint_library.filter('vehicle'))

            # This time we are using try_spawn_actor. If the spot is already
            # occupied by another object, the function will return None.
            npc = world.try_spawn_actor(bp, transform)
            if npc is not None:
                actor_list.append(npc)
                npc.set_autopilot(True)
                print('created %s' % npc.type_id)

```

DSantosO, 2021-08-13:
Hello @hailuu684,
When you spawn vehicles in the batch you can get the [returned responses and from the get all the actors' ids](https://carla.readthedocs.io/en/0.9.12/python_api/#commandresponse). With those, you can just use the [get_actor() method of world to get the reference to the python actors](https://carla.readthedocs.io/en/0.9.12/python_api/#carlaworld). 
In a general and pythonic way, I usually do something like this:
```           responses = self.client.apply_batch_sync(batch)

            veh_ids = [x.actor_id for x in responses]
            veh_refs = [self.world.get_actor(x) for x in veh_ids]

hailuu684, 2021-08-16:
Thank you so much @DSantosO. Now I can get all their information

closed 2021-08-16 by hailuu684