carla-simulator/carla #4535: B2 ERROR with "make PythonAPI" and "make launch"

ThibaultFy, 2021-08-13:
### Discussed in https://github.com/carla-simulator/carla/discussions/4534

<div type='discussions-op-text'>

<sup>Originally posted by **ThibaultFy** August 13, 2021</sup>
I'm having trouble with my Windows build for 0.9.12. I'm getting the following message:

> -[install_boost]: [B2 ERROR] An error ocurred while installing using "b2.exe". 
> -[install_boost]: Exiting with error...

Thanks for your help !</div>

ThibaultFy, 2021-08-13:
My problem solved by removing the build folder and changing the path on my repo... I don't have more explanation than this...

TrinhTuanHung2021, 2021-08-14:
> My problem solved itself by removing the build folder and changing the path on my repo... I don't have more explanation than this...
Hi
Could you guide me on how to do "make pythonAPI" command? I run it on X64 native tool command prompt but it crashes all the time. Thanks

ThibaultFy, 2021-08-16:
> Hi
> Could you guide me on how to do "make pythonAPI" command? I run it on X64 native tool command prompt but it crashes all the time. Thanks

Hi, I can't really help you from scratch, but you may find a lot of information on the Carla website, discord or github. If you have a recurrent issue, maybe you should open a specific one with detailed in order to ask for help to the carla community :)

BedirhanKeskin, 2021-10-14:
I also have the same problem with @ThibaultFy for Windows Build. Carla version is 0.9.12

BedirhanKeskin, 2021-10-20:
I was having the error when I run "make Launch". I downgraded the python version to 3.8.10 from 3.10. I reinstalled CMake, Make. Then I applied the document from Part Two: Build CARLA. Now, there is no error b2.exe error.

vini-at-potential, 2021-11-10:
Same here, had to downgrade from Python 3.10 to 3.8.10 to make it work. I did not try Python 3.9.6 though, so don't know which is the latest python version possible.

hche8927, 2021-12-23:
Same here, the issue was likely caused by Python 3.10, installing Python 3.9.9 fixed it for me.

lyjlalala, 2022-03-10:
[image]
As the screenshot shows, I got the setup error when I was typing "make PythonAPI".  
My setting:
- Windows 10 (not linux in ubuntu)
- VS: 2019 with additional elements. 
- Make: GNU Make 3.81
- CMake: 3.23.0-rc2
- Python: 3.8.8
I'm not sure what causes this error, can anyone help?

ruixuanz, 2022-03-11:
> [image] As the screenshot shows, I got the setup error when I was typing "make PythonAPI". My setting:
> 
> * carla: 0.9.13
> * Windows 10 (not linux in ubuntu)
> * VS: 2019 with additional elements.
> * Make: GNU Make 3.81
> * CMake: 3.23.0-rc2
> * Python: 3.8.8
>   I'm not sure what causes this error, can anyone help?

Same issue here. Have you found a solution?

RafesiA, 2022-09-02:
I really sorry for my bad english skill first.

I solved this issue by other way.

The bottom line is **unzip `boost_1_72_0.zip` manually and paste it into `boost-1.72.0-source'**

make launch or make PythonAPI, processes will create `carla/Build/boost_1_72_0.zip`. and expanding. When which task done, PowerSheel throw b2.exe error like this.

So i tried b2.exe --help in PowerShell, it throw error 'unable to load boost.build: could not find "boost-build.jam"'

I guess it should be missing files when expanding on PowerShell. and checked downloaded zip file, compare with unziped folder. finally i found some files are missed while exapnding on PowerShell.

That's it. i just unzip `boost_1_72_0.zip` manually, and it does work.

Ja20019, 2022-09-12:
this issue was caused by Python 3.10, to remove python 3.10 then install  python 3.8 to fix it.

T1murY, 2024-11-06:
> I'm having the same issue. I've followed the latest docs for building in Windows, but this error keeps occurring: ![螢幕擷取畫面 2024-10-30 081358](https://private-user-images.githubusercontent.com/100141401/381451647-0e477781-e356-4758-95d7-e45ac0530672.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA5MTkyODMsIm5iZiI6MTczMDkxODk4MywicGF0aCI6Ii8xMDAxNDE0MDEvMzgxNDUxNjQ3LTBlNDc3NzgxLWUzNTYtNDc1OC05NWQ3LWU0NWFjMDUzMDY3Mi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEwNlQxODQ5NDNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05Zjc1ODg4MmIwZDZiOGE0NzVjY2UwZGJmZTcyZTQxNDRhNTYwNjZlNTQ0YjhlN2Q3ZjQyNWZkZGYyMjY2OGNlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.zusAwu4G7MxsWKD7eDkJ-jfV6cVXQVbh_cNAGxNw6VI) My setting:
> 
>     * carla: 0.9.15
> 
>     * Windows 11
> 
>     * VS: 2022 with additional elements.
> 
>     * Make: GNU Make 3.81
> 
>     * CMake: 3.30.2
> 
>     * Python: 3.8.0
> 
> 
> I already try unzip boost_1_80_0.zip manually and excute bootstrap.bat , but it doesn't work I'm not sure what causes this error, can anyone help?

I have this issue as well exactly the same, ı have followed this https://github.com/carla-simulator/carla/issues/7082#issuecomment-2049319061 issue to here and stuck here any solution?
[image]

Siddhartha80, 2024-11-23:
I got an error while excuting make PythonAPI

 -[install_boost]: [B2 ERROR] An error ocurred while installing using "b2.exe".
    -[install_boost]: Exiting with error...
    
    
  Please guide me how do I solve it

Shunqiang-Feng, 2024-12-10:
Hi friends,

I'm using python-3.8.10, I'm sure that I only have this version of python in my computer. However, **B2.exe** erros also showed in my pc.

My solution [ from ID "Mohammad issam" in _discord.com_]:

1. It seems that B2.exe is used to deal with boost file. So, if we can extract the boost files in carla's build folder, this problem might be fixed.
2. Here is boost file:
[boost-1.80.0-install.zip](https://github.com/user-attachments/files/18078351/boost-1.80.0-install.zip)

[boost-1.80.0-source.zip](https://github.com/user-attachments/files/18078352/boost-1.80.0-source.zip)
3. Notice: Ensure that you extract in "carla-0.9.15/Build/" and check the parent folder relationship after unzip.

nuralmmdv, 2024-12-23:
@Shunqiang-Feng thanks bro! its proposal fixed this issue in my side.

Shunqiang-Feng, 2024-12-23:
@nuralmmdv  you're welcome. BTW, to be honest, I finally chose another building option. In my platform [win11, VS 2022], problems came one after another. Now, [this option](https://carla-ue5.readthedocs.io/en/latest/build_windows_ue5/) works for me. But I have to say that you may need to modify the plugin (for example, chose the win11 version instead of win10, and balabala...)
Ok, that's all. Hope everything goes well.

nuralmmdv, 2024-12-24:
@Shunqiang-Feng yep you are right. during installation, this issue was fixed with your way but after proceeding installation, I had several stubborn problems and I got stuck at the end :D now I am following this video, I will update this message if the installation is done well.

https://www.youtube.com/watch?v=wdeHxZKWaIc&t=305s

nuralmmdv, 2024-12-24:
final situation :D guys ignore the previous comment. Follow these video series and you will be install successfully.

https://www.youtube.com/watch?v=7vFy3aanwwo

sdhar16, 2025-03-27:
Hi people, I debugged the issue with VS 2022. Boost 1.80's bootstrap file call tools\build\src\engine\build.bat and uses the latest vc toolkit, it does not pick up the correct toolkit, in my case vc143, but Carla install script calls bootstrap.bat with vc141 and b2 build with vc142. This causes the build to fail. 

My recommendation is to change the build to vc143 in all calls in Util\BuildTools\Windows.mk line 75, and Util\InstallersWin\install_boost.bat in line 109. 

It should work after that.

Blyron, 2025-05-20:
We have upgraded the default build system of UE4 to VS2022 closing this issue as it was related with it

closed 2025-05-20 by Blyron