carla-simulator/carla #4536: [possible bug] - Collision Sensor

LucasFKobernic, 2021-08-13:
Hi everyone, 

I am experiencing strange behavior from the "collision sensor". I am simulating a frontal collision between a pedestrian and the vehicle at different points on the front of the vehicle. (Windows 10, Carla 0.9.12)

Whenever the collision point is >=50% of the center of the vehicle, the sensor does not register the collision. The most interesting thing is that if you change the pedestrian direction (from left instead of right) so that what was >50% Width before is now <50%, it recognizes the collision, but what is now > 50% he no longer recognizes.
[image]
[image]

I recorded this strange sensor behavior with different vehicles.

Does anyone have any idea what it might be? Or is it a bug in the collision sensor?

glopezdiest, 2021-12-01:
That definitely seems weird. I'll take a look see if I can reproduce the issue

LucasFKobernic, 2021-12-09:
Hi @glopezdiest thanks for the reply.

I tried t improve the collision detection inside of Unreal Engine changing some parameters, but i was unsucessful. 

If you want, i can share with you the code to simulate this strange behavior.

i really appreciate any help here

glopezdiest, 2021-12-10:
Yeah sure any code you provide is highly appreciated

glopezdiest, 2021-12-20:
@LucasFKobernic From what I'm seeing, I tried to collide multiple times with both static and dynamic pedestrian from different directions and they all seem to be working fine. So I guess I need more details on your use-case:
1) Give me one vehicle and walker with which that happens
2) What is the vehicle doing? Is it static, accelerating...?
3) What is the pedestrian doing? Is it static? How is it controlled?

LucasFKobernic, 2022-01-10:
Hi @glopezdiest,

Sorry for the delayed response. Here the answers to your questions:

1. Im using walker.pedestrian.0001 and  vehicle.seat.leon'. but the problem also happens with other pedestrians and vehicles.
2. The vehicle is just moving forward. Im using "enable_constant_velocity" and i have tried different velocities going from 10 km/h to 60 km/h and it seems to no have any influence on that.
3. The pedestrian just cross the street. 

I made a example for you:
`def main():
    argparser = argparse.ArgumentParser(
        description=__doc__)
    argparser.add_argument(
        '--sync',
        action='store_true',
        help='Synchronous mode execution')
    argparser.add_argument(
        '-s', '--seed',
        metavar='S',
        type=int,
        help='Random device seed')
    argparser.add_argument(
        '--hybrid',
        action='store_true',
        help='Enanble')
    argparser.add_argument(
        '--velocityv',
        metavar='VV',
        default=10,
        type=float,
        help='Vehicle s velocity. Defaul 10 km/h ')
    args = argparser.parse_args()
    logging.basicConfig(format='%(levelname)s: %(message)s', level=logging.INFO)

    # ---
    # -- Connecting the client
    # ---
    client = carla.Client('127.0.0.1', 2000)
    client.set_timeout(10.0)
    synchronous_master = False
    random.seed(args.seed if args.seed is not None else int(time.time()))

    # -- Arrays to save simulation Information
    actors_list = []
    global COLLISION
    COLLISION = False
    try:
        world = client.get_world()
        traffic_manager = client.get_trafficmanager(8000)
        if args.hybrid:
            traffic_manager.set_hybrid_physics_mode(True)
        if args.seed is not None:
            traffic_manager.set_random_device_seed(args.seed)

        if args.sync:
            mode = 'sync'
            settings = world.get_settings()
            traffic_manager.set_synchronous_mode(True)
            if not settings.synchronous_mode:
                synchronous_master = True
                settings.synchronous_mode = True
                settings.fixed_delta_seconds = 0.05
                world.apply_settings(settings)
            else:
                synchronous_master = False
        else:
            mode = 'async'

        # ---
        # Pedestrian Spawn Point
[... 87 lines cut ...]
        print('\nAll actors destroyed')
        if args.sync and synchronous_master:
            settings = world.get_settings()
            settings.synchronous_mode = False
            settings.fixed_delta_seconds = None
            world.apply_settings(settings)

if __name__ == '__main__':
    try:
        main()
    except KeyboardInterrupt:
        pass
    finally:
        print('done.')`
Changing at line 221 the condition for the pedestrian you can see the strange behavior:

1. setting it to 11,5 the pedestrian will colide with the car after the middle but the collision sensor will not get it. So the simulation continues running. 

https://user-images.githubusercontent.com/77980399/148749854-848466fa-12b9-44ef-acdd-72e922eca641.mp4

2. setting it to 10 the pedestrian will colide with the vehicle before the middle, the collision_sensor will reconigze it and the simulation will stop.

https://user-images.githubusercontent.com/77980399/148750177-97f84a42-3d99-499d-bf70-b648759b474e.mp4

I hope these Information can help u to reproduce the error :)

glopezdiest, 2022-01-10:
Okay, I'll take a look. Also, I edited your comment for better visibility

glopezdiest, 2022-01-10:
You seem to have a `collision_callback` function, which isn't part of that code + what arguments are you using?

LucasFKobernic, 2022-01-13:
Hi @glopezdiest 

here the function:

`def collision_callback(sensor_data):
    global COLLISION
    if sensor_data.other_actor is not None:
        COLLISION = True`

I also added a UE_Log in the cpp code, to always get a warning message when the collision sensor detects a collision. For the cases where the collision is not being detect, I'm getting no warning message. So I think the problem is not on the client side, but maybe at Unreal Engine?

glopezdiest, 2022-01-17:
After some tests, I can easily reproduce the error. That's definitely an issue. The collision sensor seems to be working fine, and the problem seems to be with the collision not triggered (hence there is nothing to catch by the sensor). However, this is only happening for invincible pedestrian, and if you take their invincibility out, they work just fine:

```py
if walker_bp.has_attribute('is_invincible'):
    walker_bp.set_attribute('is_invincible', 'false')  
```

Adding it to our to do list, we'll see when we can fix it. Thanks for everything :slightly_smiling_face:

LucasFKobernic, 2022-01-18:
Hi @glopezdiest 

I have test it and now it works. Thank you for your help!

closed 2022-01-18 by LucasFKobernic