carla-simulator/carla #4537: Changes in license plate blueprint has not effect

ebadi, 2021-08-13:
CARLA version: carla-0.9.12
Platform/OS: Ubuntu 18.04.5
Problem you have experienced: 
I followed the instruction in this link https://github.com/carla-simulator/carla/issues/3429 by @Madecu but I see no change in the license plates. 
Here is the video that shows my modification and its result when I run `generate_traffic.py` script. 

https://www.youtube.com/watch?v=pk6hzE6O_tI

Madecu, 2021-08-13:
Hi @ebadi You don't have these nodes connected

[image]

ebadi, 2021-08-13:
Thanks @Madecu  for pointing out my mistake. You suggestion fixed the licence plate issue for **few of** the vehicles (one vehicle at second 14, https://youtu.be/laMCjPqfd64?t=14). As you see in the rest of the video, most of the vehicles are still driving around with the default license plate.
Would you like me to close this issue and do you want to change the label of this issue to "bug report" ?
Thanks again for the great work on Carla project.

ebadi, 2021-08-16:
More information about this bug, only the following first generation vehicles support dynamic license plates: 

- vehicle.tesla.model3
- vehicle.volkswagen.t2
- vehicle.lincoln.mkz_2017
- vehicle.mini.cooper_s
- vehicle.ford.mustang
- vehicle.audi.tt

gsmario, 2021-08-17:
Hello @ebadi.
In addition to those changes Madecu told you, you also need to change the slot name in the vehicle material, as shown in the image. You need to use the name "LicensePlate_Mat" that way, the blueprint knows which one is the licenseplate material and it will change it.
We don't use this feature, thats why its deactivated in carla by default, but if you follow those steps, it will work!
Let me know if you have any problems.

[image]

ebadi, 2021-08-19:
Thanks @gsmario  
This solved my issue.

closed 2021-08-19 by ebadi