carla-simulator/carla #4540: Support multi-threading in traffic manager

qhaas, 2021-08-15:
Feature Request relating to Issue #3442 and #4289

CARLA version: **0.9.12**
Platform/OS: Linux **(Ubuntu 18.04/20.04+ RHEL 7/8)**
Problem you have experienced: **Traffic Manager appears to only use a single core and causes CPU bottlenecks with many NPCs**
What you expected to happen: **CARLA PythonAPI client's python process uses more than one core on a multi-core system**
Steps to reproduce: **Run a simulation with dozens of vehicles with auto_pilot enabled, watch `top`, the CPU usage doesn't go above 100%**

Thought I'd [`Use the updated Traffic Manager to populate the map with hundreds of cars`](https://carla.org/2021/08/02/release-0.9.12/), but I quickly became CPU bottle-necked despite having plenty of idle cores.

Per the [CARLA Performance Report](https://carla.readthedocs.io/en/latest/adv_benchmarking/#carla-performance-report), and my own testing, the more vehicles with auto pilot enabled, the slower the simulation runs, to the point where it is nearly unusable after a few dozen vehicles are added.  From observing the pythonapi client's cpu utilization, it appears that the traffic manager is single threaded since it never uses more than 100% CPU even on a multi-core system where other (multi-threaded) processes commonly use +200%.

jackbart94, 2021-08-20:
Thanks for pointing this out to us! We've been doing a lot of profiling during the development of 0.9.12 (and also already merged part of the PR you mentioned) and even though Traffic Manager would benefit from multi-threading, there are other bottlenecks that are involved. Anyway, we'll be doing more profiling surely while developing 0.9.13, and we'll try to improve the performance of Traffic Manager as we go!

qhaas, 2021-09-13:
Thanks for the update, glad the team is looking into it.

PS @SanggunLee [might be on to something](https://github.com/carla-simulator/carla/discussions/4556), but I haven't had a chance to try his optimization and benchmark it yet.

SanggunLee, 2021-09-13:
> Thanks for the update, glad the team is looking into it.
> 
> PS @SanggunLee [might be on to something](https://github.com/carla-simulator/carla/discussions/4556), but I haven't had a chance to try his optimization and benchmark it yet.

Hi.
My solution is for server side.
Traffic manager is client side and I think Traffic manager needs Multi thread architecture.
I know the problem of concurrency problem, so it will be hard to develop.

thanks for your contribution.

closed 2021-11-17 by germanros1987