carla-simulator/carla #4542: Fog doesn't work - mosaic images

zengtaiping, 2021-08-16:
I'm trying to create a fog environment using the following configuration:
```
carla.WeatherParameters(cloudiness=0,
                        precipitation=0,
                        precipitation_deposits=0,
                        fog_density=50,
                        fog_distance=10,
                        sun_altitude_angle=-5)
```
However, the image from the camera is with mosaic:
[image]
What can be the problem? Is it the wrong configurations?

XGodina, 2021-08-19:
@zengtaiping Could you provide me with some more information? What package is it? 9.12? The camera resolution is very low, isn't it?

zengtaiping, 2021-08-19:
@XGodina Thanks for your reply. I am sorry for the incomplete information. I used the latest version. 
And I used the following camera parameters:
```
camera = world.get_blueprint_library().find('sensor.camera.rgb')
camera.set_attribute('image_size_x', '1280')
camera.set_attribute('image_size_y', '720')
camera.set_attribute('fov', '110')
camera.set_attribute('sensor_tick', '0.02')
```

Madecu, 2021-09-07:
Hi @zengtaiping,
Sorry for the delayed respone. I've tried your weather configuration in Town01 and it works fine.

[image]

Are you sure it is the fog who make the artifact? It only happens if you set the fog? Or it works with different amount of fog?

zengtaiping, 2021-09-08:
Hi @Madecu, thanks for your reply. I tried different weathers, and only the fog one has this mosaic. And I found that more amount of fog makes it worse.

Madecu, 2021-09-09:
I'm not able to reproduce those artifacts. I tried in windows and Linux, different towns but nothing. Can you try using the environment.py script that we have in PythonAPI/util to change the fog amount and see if it still causing that issue?
Maybe it is something related with your graphic card? I don't know

Let's keep trying different things so we can discard potential causes and figure out what is causing it

zengtaiping, 2021-09-10:
@Madecu , I changed the fog amount, and it still causes this issue. I used this command `lspci | grep VGA`, and found that my gpu is `NVIDIA Corporation Device 2230 (rev a1)`. My driver version is `Driver Version: 470.63.01`. If you suddenly think of some commands I can try, please let me know. Thanks.

Madecu, 2021-09-10:
@zengtaiping, Are you using the package verrion or editing unreal's project? Which version of Unreal Engine do you have installed? What graphic card do you have?

Thank you.

zengtaiping, 2021-09-16:
@Madecu No, I did not edit the project. I use the NVIDIA A6000.

Axel1092, 2021-09-21:
@zengtaiping Are you working with ubuntu or windows? Also, Are you using the package (if so what version) or are you compiling from source?
With the server running, could you run the `test_connection.py` script in the `PythonAPI/util` folder and report the output please?

closed 2021-11-08 by zengtaiping