carla-simulator/carla #4543: Can not git clone CarlaUnreal from UnrealEngine

ethanchiang6516, 2021-08-16:
Dear sir:

Environment : Ubuntu 18.04
Can not run this order 
git clone --depth 1 -b carla https://github.com/CarlaUnreal/UnrealEngine.git ~/UnrealEngine_4.26

because git hub need "Two-factor authentication"
Ubuntu show error as follow:
remote: Support for password authentication was removed on August 13, 2021. Please use a personal access token instead.
remote: Please see https://github.blog/2020-12-15-token-authentication-requirements-for-git-operations/ for more information.
fatal: unable to access 'https://github.com/CarlaUnreal/UnrealEngine.git/': The requested URL returned error: 403

Ethan
-->

iamweiweishi, 2021-08-18:
The same issue.

CaptnMorgan833, 2021-08-31:
You have to create a personal token in your GitHub Account settings (under developer settings). See https://github.blog/2020-12-15-token-authentication-requirements-for-git-operations/ as mentioned in your error log.

Then it worked.

But I ran into the next problem. After setting up the personal token I get the following message when cloning:

fatal: Repository 'https://github.com/CarlaUnreal/UnrealEngine.git/' not found.

It seems the repository does not excist anymore. Or has it been moved to another repository?

CaptnMorgan833, 2021-08-31:
Ah okay, did not accept the invitation from EpicGames after connecting EpicGames acc to my GitHub acc. This is why I had no access to the repository.

closed 2021-12-01 by MattRowe18

zlg9folira, 2022-05-29:
I have the same issue, already accepted the invitation link and received confirmation email. Still get this error when using my token:
```unable to access 'https://github.com/CarlaUnreal/UnrealEngine.git/': The requested URL returned error: 403```

bipul-mohanto, 2022-07-21:
Same Issue Also.

I am using Ubuntu 22.04, and trying to work on unreal engine 4.0 version as this [blogpost](https://www.addictivetips.com/ubuntu-linux-tips/unreal-engine-4-ubuntu/) shows. I have generated my `github personal access token` already, and my github account is also connected with `epic games`. I have used the command `git clone --recursive --branch release https://github.com/EpicGames/UnrealEngine.git`. However, this is returning this error:

```ruby
remote: Write access to repository not granted.
fatal: unable to access 'https://github.com/EpicGames/UnrealEngine.git/': The requested URL returned error: 403

```

* did anyone has the solution? 
* I actually want to work with the path tracing, and found unreal 4.27 or latest does not support Ubuntu. What is the latest version I can install on Ubuntu (4.0?)

gitHql, 2022-12-05:
the documents was out of date for the elder versions, the link becomes :
    git clone --depth 1 -b carla https://github.com/CarlaUnreal/UnrealEngine.git ~/UnrealEngine_4.26

when you choose the latest version.
I have fork it and clone to local, seemed alright

Shesh02, 2023-03-27:
I have added the ssh.pub generated keys to github account still i'm getting this below error . 

$ git clone --depth 1 -b carla https://github.com/CarlaUnreal/UnrealEngine.git
Cloning into 'UnrealEngine'...
remote: Repository not found.
fatal: repository 'https://github.com/CarlaUnreal/UnrealEngine.git/' not found

kashwp7, 2023-04-23:
Try the following if it still not working after the invitation is accepted.

git clone --depth 1 -b carla git@github.com:CarlaUnreal/UnrealEngine.git ~/UnrealEngine_4.26

the https URL doesn't seem to work like the documentation says

MattRowe18, 2023-04-23:
@kashwp7 this problem normally arises when you haven't successfully connected your GitHub account to Epic Games. Try following those steps completely again. Git doesn't report the error as an authentication error, it claims the URL is not found.

accidentalgamesstudio, 2023-05-21:
For those still struggeling, try using the Personal access tokens (classic) rather than Fine-grained personal access tokens when generating your token in Github.

bapukokare, 2023-07-26:
I directly download it into local directory with required UE4 version, follow this process 
https://www.unrealengine.com/en-US/ue-on-github

2019211753, 2023-08-01:
> [accidentalgamesstudio](/accidentalgamesstudio)

great! u are right!

sauvikgon, 2023-12-29:
> Same Issue Also.
>
> I am using Ubuntu 22.04, and trying to work on unreal engine 4.0 version as this [blogpost](https://www.addictivetips.com/ubuntu-linux-tips/unreal-engine-4-ubuntu/) shows. I have [...]

You have to create a Token(Classic) instead of the Fine Grained Token. Then you will be able to clone the repo. 
Man I just found that out after wasting one full day on this thing.

samehmohamed88, 2024-02-09:
> git clone --depth 1 -b carla https://github.com/CarlaUnreal/UnrealEngine.git ~/UnrealEngine_4.26

I can confirm also that @accidentalgamesstudio solution is correct.  Hope this helps others

Jubin2607, 2024-03-01:
Still the same issue, Can someone suggest a solution

luohuan2uestc, 2024-03-25:
> Still the same issue, Can someone suggest a solution
Generate a Personal Access Token (PAT): Log in to your GitHub account, go to Settings > Developer settings > Personal access tokens, and click on "Generate new token."

Set token permissions: During the token generation process, you will need to select the permissions for the token. To clone a repository, you will at least need repo permissions to access your private repositories.

Use the token for authentication: After generating the token, you can use it when cloning the repository. For example, if your token is ghp_12345abcdefg, you can use it in the clone command like this:

git clone https://oauth2:ghp_12345abcdefg@github.com/CarlaUnreal/UnrealEngine.git

anshgwash, 2024-06-11:
Tried all the above but still getting the same issue: `fatal: repository 'https://github.com/CarlaUnreal/UnrealEngine.git/' not found`

rmrtcla, 2024-06-17:
> Tried all the above but still getting the same issue: `fatal: repository 'https://github.com/CarlaUnreal/UnrealEngine.git/' not found`

It could be your problem is similar to the ones mentioned in https://github.com/carla-simulator/carla/issues/7788 and https://github.com/carla-simulator/carla/issues/7775 . If so, then there's nothing you can do right now except monitor these issues here and also https://forums.unrealengine.com/t/maintenance-on-the-epic-games-github-organization-june-2024/1891135

rmrtcla, 2024-06-17:
@MattRoweEAIF just FYI, this hole Epic Unreal sign-up process seems broken right now for new accounts since June 10th due to organisational changes on the Epic Github Org level. There are many different repos and people affected, see the link I posted before. It all _seems_ to work, but the CarlaUnreal repo remaines 404 no matter what you try...

michaelraab, 2024-07-02:
Has someone found a solution for this issue?

966007, 2024-10-12:
> 也是同样的问题。
> 
> 我正在使用 Ubuntu 22.04，并尝试在虚幻引擎 4.0 版本上工作，如这[篇博文](https://www.addictivetips.com/ubuntu-linux-tips/unreal-engine-4-ubuntu/)所示。我已经生成了我的，我的 github 帐户也与 连接了。我使用了命令 。但是，这将返回此错误：`github personal access token``epic games``git clone --recursive --branch release https://github.com/EpicGames/UnrealEngine.git`
> 
> ```ruby
> remote: Write access to repository not granted.
> fatal: unable to access 'https://github.com/EpicGames/UnrealEngine.git/': The requested URL returned error: 403
> ```
> 
> * 有人有解决方案吗？
> * 我其实想用路径追踪，发现虚幻的 4.27 或最新版本不支持 Ubuntu。我可以在 Ubuntu 上安装的最新版本是什么 （4.0？）

zwj-sep, 2025-03-13:
> 我有同样的问题，已经接受了邀请链接并收到了确认电子邮件。使用我的令牌时仍然出现此错误： `unable to access 'https://github.com/CarlaUnreal/UnrealEngine.git/': The requested URL returned error: 403`

您好 请问您后来怎么解决的呀