carla-simulator/carla #4551: CARLA server can't start

ReCodeLife, 2021-08-18:
**CARLA version**:0.9.11
**Platform/OS**:wsl2 Ubuntu18.04
**Problem you have experienced:**
   I'm using docker to build the environment, but my carla service can't start, which means that when I run './CarlaUE4.sh', the console only outputs:
"4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps."
   The window didn't appear, the memory didn't take up, and it was over. I don't know which step went wrong, maybe it was the docker's problem, maybe it was wsl's problem.
   If it's a carla version issue, which version of the image I should choose would be better

**What you expected to happen:**
A window pops up.

**Steps to reproduce:**
docker run \
  -e SDL_VIDEODRIVER=x11 \
  -e DISPLAY=$DISPLAY \
  -v /tmp/.X11-unix:/tmp/.X11-unix \
  -p 2000-2002:2000-2002 \
  -it \
  --gpus all \
 carlasim/carla:0.9.11 bash

and then:
./CarlaUE4.sh

So what do I do?

nuomizai, 2021-08-19:
First, does your environment has a usable CUDA device and driver?

If it does, maybe you can try the following command to run carla in an offscreen mode
```
 SDL_VIDEODRIVER=offscreen ./CarlaUE4.sh -quality-level=Epic -resx=800 -resy=600 -opengl
```
I haven't run a carla server on a WSL ubuntu server, but I have run the carla server on a headless ubuntu server successfully using the following steps. First, I used the following command to install virtualgl on a ubuntu server without visual interface
```
sudo apt-get install xorg libxcb-randr0-dev libxrender-dev libxkbcommon-dev libxkbcommon-x11-0 libavcodec-dev libavformat-dev libswscale-dev
sudo nvidia-xconfig -a --use-display-device=None --virtual=1280x1024
https://sourceforge.net/projects/virtualgl/files/2.5.2/virtualgl_2.5.2_amd64.deb/download -O virtualgl_2.5.2_amd64.deb
sudo dpkg -i virtualgl*.deb
rm virtualgl*.deb
sudo reboot
```
Then, check your nvidia driver by 
```
nvidia-smi
```
If not, use the following command to install a proper driver
```
sudo ubuntu-drivers devices
sudo ubuntu-drivers install
```
After that, start X display by
```
nohup sudo X
export DISPLAY=:0.0
```
Last, try the command to run a headless carla server
```
SDL_VIDEODRIVER=offscreen ./CarlaUE4.sh -quality-level=Epic -resx=800 -resy=600 -opengl
```
This method works for me in a headless ubuntu server.

ReCodeLife, 2021-08-19:
> First, does your environment has a usable CUDA device and driver?
>
> If it does, maybe you can try the following command to run carla in an offscreen mode
>
> ```
>  SDL_VIDEODRIVER=offscreen ./CarlaUE4.sh [...]

I'm using docker, and I think maybe it's already loaded with drivers, nvidia-smi has an output, but I can't run” nvidia-xconfig“， it shows:" command not found". 
Maybe you have contact information, qq or WeChat, I would like to ask for your help.

qhaas, 2021-08-24:
I use `xmessage` to make sure X11 is setup properly inside a container and `glxgears` to test if opengl rendering is working correctly inside a container.  xmessage / glxgears is much less complex than UnrealEngine / Carla when troubleshooting.  If neither of those work, you will not have luck getting carla to work.

FWIW, CARLA [recently updated the documentation for running in docker](https://carla.readthedocs.io/en/0.9.12/build_docker/#off-screen-mode), but doing so on top of WSL2 with GPU acceleration is likely uncharted territory

ReCodeLife, 2021-08-25:
> I use `xmessage` to make sure X11 is setup properly inside a container and `glxgears` to test if opengl rendering is working correctly inside a container. xmessage / glxgears is much less complex than UnrealEngine / Carla when troubleshooting. If neither of those work, you will not have luck getting carla (with a GUI) to work.
> 
> FWIW, CARLA [recently updated the documentation for running in docker](https://carla.readthedocs.io/en/0.9.12/build_docker/#off-screen-mode), but doing so on top of WSL2 with GPU acceleration is likely uncharted territory

> I use `xmessage` to make sure X11 is setup properly inside a container and `glxgears` to test if opengl rendering is working correctly inside a container. xmessage / glxgears is much less complex than UnrealEngine / Carla when troubleshooting. If neither of those work, you will not have luck getting carla (with a GUI) to work.
> 
> FWIW, CARLA [recently updated the documentation for running in docker](https://carla.readthedocs.io/en/0.9.12/build_docker/#off-screen-mode), but doing so on top of WSL2 with GPU acceleration is likely uncharted territory

I have looked up a lot of information, maybe WSL can not install the driver, so it cannot run. Now THAT I have installed the native Ubuntu system, I can easily use Docker to build Carla. It's ready to use now. Thank you very much anyway.

closed 2021-08-25 by ReCodeLife

WangTianYuan, 2021-09-02:
@ReCodeLife 你好，我现在也是在docker上安装的carla，出现了和你一样的问题，请问你是怎么解决的？我的服务器是没有显示器的，用官方的offscreen mode不好使

ReCodeLife, 2021-09-02:
> @ReCodeLife 你好，我现在也是在docker上安装的carla，出现了和你一样的问题，请问你是怎么解决的？我的服务器是没有显示器的，用官方的offscreen mode不好使

我使用的是ubuntu桌面版，但是理论上来说你开个vnc服务，将display映射到vnc端口上，也是可行的。

ReCodeLife, 2021-09-08:
> @ReCodeLife 可以冒昧的加一下你的微信吗？我的是wangtianyuan1111，我刚学习使用carla，可能会有一些问题要请教你，感激不尽

最近比较忙 刚看到 不知道你是否还需要帮助

WangTianYuan, 2021-09-09:
@ReCodeLife 我还是没有安装好，如果能够帮忙感激不尽。我直接docker pull carla还是用不了，应该是UE4没法启动，运行./CarlaUE4.sh 只显示core dumps，然后就回到命令行了，我找不到错误的原因，我们这边目前有一台机器是在linux原生环境下装的，但是docker上一直不行，请问你有额外装什么东西吗？

AOOOOOA, 2023-11-28:
> @WangTianYuan Hello, I ran into a similar problem. But I am not running it on docker, instead, I am trying to run Carla in a server without display and root permission. Could you please share any thought?
> 
> [image] Only those two lines showed up, then crash.
> 
> I am using 0.9.13, with this command: "./CarlaUE4.sh -RenderOffScreen"

I met the same error. However, I have tried on several servers(ubuntu), only one server has this problem. I don't know why. Have you solved this problem? If yes, how? Thanks.

ChipsICU, 2024-08-21:
> I met the same error. However, I have tried on several servers(ubuntu), only one server has this problem. I don't know why. Have you solved this problem? If yes, how? Thanks.

Did you fix this problem? I do have the same problem, qaq.