carla-simulator/carla #4553: RuntimeError: Responding error from function get_light_boxes: Actor could not be found in the registry

nuomizai, 2021-08-18:
When I ran the code `light_box = _actor.get_light_boxes()`, an error occurred as 
```   
RuntimeError: Responding error from function get_light_boxes: Actor could not be found in the registry.  Actor Id: 370
```
The _actor was a traffic light actor which is obtained by `CarlaDataProvider.get_next_traffic_light(self._vehicle)` in leaderboard project. The detailed implementation of `get_next_traffic_light` is as follows:

```
@staticmethod
    def get_next_traffic_light(actor, use_cached_location=True):
        """
        returns the next relevant traffic light for the provided actor
        """

        if not use_cached_location:
            location = actor.get_transform().location
        else:
            location = CarlaDataProvider.get_location(actor)

        waypoint = CarlaDataProvider.get_map().get_waypoint(location)
        # Create list of all waypoints until next intersection
        list_of_waypoints = []
        while waypoint and not waypoint.is_intersection:
            list_of_waypoints.append(waypoint)
            waypoint = waypoint.next(2.0)[0]

        # If the list is empty, the actor is in an intersection
        if not list_of_waypoints:
            return None

        relevant_traffic_light = None
        distance_to_relevant_traffic_light = float("inf")

        for traffic_light in CarlaDataProvider._traffic_light_map:
            if hasattr(traffic_light, 'trigger_volume'):
                tl_t = CarlaDataProvider._traffic_light_map[traffic_light]
                transformed_tv = tl_t.transform(traffic_light.trigger_volume.location)
                distance = carla.Location(transformed_tv).distance(list_of_waypoints[-1].transform.location)

                if distance < distance_to_relevant_traffic_light:
                    relevant_traffic_light = traffic_light
                    distance_to_relevant_traffic_light = distance

        return relevant_traffic_light
```

Does anyone know where the problem is?

chingolo7278, 2021-09-09:
I have the same issue, I can't use the function get_light_boxes
"RuntimeError: Responding error from function get_light_boxes: Actor could not be found in the registry.  Actor Id: 16"
Does anyone has an example of this function?

KevinLADLee, 2021-11-15:
I think a conditional sentence should be added here.

https://github.com/carla-simulator/carla/blob/bb4b95e0bc43d25be6f3b54713ac32d8465b1586/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Server/CarlaServer.cpp#L1645

```cpp
if (CarlaActor == nullptr)
```
After modifying this line, the function works now.

100rab-S, 2021-12-03:
I have the same issue. What is the solution for this? Other methods work on the trafficLight object. It is issue only with get_light_boxes.

lsangreg, 2023-01-24:
@Axel1092  Do you have any clue on that issue? Is still unsolved for me, with carla 0.9.12

```
        actor_list = self.world.get_actors()

        traffic_light_list= actor_list.filter('traffic.traffic_light*')
        for tl in traffic_light_list:
            
            try:
                light_box=tl.get_light_boxes()
                print(light_box)
            except RuntimeError:
                print("Actor not found")
                pass
            
 ```
The function is not working for each traffic light in the map.