carla-simulator/carla #4554: At crossing road with auto mode, car will drive with S line

DanielTian90, 2021-08-19:
Like the title, while run my code at crossing road with auto mode, car will drive with S line or drive with cycle. So the reason is?
See gif.

jackbart94, 2021-08-20:
Please, give more information about which script you are using, which version of CARLA, and a code snippet to reproduce the error. From what I can understand, you're using asynchronous mode. It's recommended to use synchronous mode if you're using the autopilot. If you're using ``manual_control.py`` it means that you need to add the ``--sync`` flag.

DanielTian90, 2021-08-24:
My script is a combination of automatic_control.py and addding spectator function and radar output
Carla version: 0.9.11
#!/usr/bin/env python
Copyright (c) 2018 Intel Labs.
authors: German Ros (german.ros@intel.com)
This work is licensed under the terms of the MIT license.
For a copy, see https://opensource.org/licenses/MIT.

"""Example of automatic vehicle control from client side."""

from future import print_function

import argparse
import collections
import datetime
import glob
import logging
import math
import os
import random
import re
import sys
import weakref
import time
try:
import pygame
from pygame.locals import KMOD_CTRL
from pygame.locals import K_ESCAPE
from pygame.locals import K_q
except ImportError:
raise RuntimeError('cannot import pygame, make sure pygame package is installed')

try:
import numpy as np
except ImportError:
raise RuntimeError(
'cannot import numpy, make sure numpy package is installed')
==============================================================================
-- Find CARLA module ---------------------------------------------------------
==============================================================================

try:
sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
sys.version_info.major,
sys.version_info.minor,
'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
pass
==============================================================================
-- Add PythonAPI for release mode --------------------------------------------
==============================================================================

try:
sys.path.append(os.path.dirname(os.path.dirname(os.path.abspath(file))) + '/carla')
except IndexError:
pass

import carla
from carla import ColorConverter as cc

from agents.navigation.behavior_agent import BehaviorAgent # pylint: disable=import-error
from agents.navigation.roaming_agent import RoamingAgent # pylint: disable=import-error
from agents.navigation.basic_agent import BasicAgent # pylint: disable=import-error
==============================================================================
-- Global functions ----------------------------------------------------------
==============================================================================

def find_weather_presets():
"""Method to find weather presets"""
rgx = re.compile('.+?(?:(?<=[a-z])(?=[A-Z])|(?<=[A-Z])(?=[A-Z][a-z])|$)')
[... 816 lines cut ...]
argparser.add_argument(
    '-s', '--seed',
    help='Set seed for repeating executions (default: None)',
    default=None,
    type=int)

args = argparser.parse_args()

args.width, args.height = [int(x) for x in args.res.split('x')]

log_level = logging.DEBUG if args.debug else logging.INFO
logging.basicConfig(format='%(levelname)s: %(message)s', level=log_level)

logging.info('listening to server %s:%s', args.host, args.port)

print(__doc__)

try:
    game_loop(args)

except KeyboardInterrupt:
    print('\nCancelled by user. Bye!')

if name == 'main':
main()

closed 2022-05-02 by stale[bot]