carla-simulator/carla #4560: 0.9.12: "-opengl" not supported anymore

guntherkrehl, 2021-08-19:
Hi there,

CARLA version: 0.9.12
Platform/OS: Unbuntu 18.04

Problem you have experienced:
"./CarlaUE4.sh -opengl"
Shuts down right away with a message "Warning: OpenGL is no longer supported for desktop platforms. Vulkan will be used instead.", which is fine if you want the windowed mode but it crashes if you use it in hthe headless mode like:
"SDL_VIDEODRIVER=offscreen ./CarlaUE4.sh -opengl"
Because Vulkan dies.

What you expected to happen:
"SDL_VIDEODRIVER=offscreen ./CarlaUE4.sh -opengl" was always used to start headless / in the background without a spectator window. Is there an updated way how to do this?

Steps to reproduce:
"./CarlaUE4.sh -opengl"
and
"SDL_VIDEODRIVER=offscreen ./CarlaUE4.sh -opengl"

Thanks for your help!

qhaas, 2021-08-24:
Epic deprecated 'desktop' OpenGL in [UnrealEngine 4.23](https://docs.unrealengine.com/4.26/en-US/WhatsNew/Builds/ReleaseNotes/4_23/) and [disabled it in UE 4.26](https://docs.unrealengine.com/4.26/en-US/WhatsNew/Builds/ReleaseNotes/4_26/), this is an upstream UnrealEngine thing, unfortunately.  

I tried `SDL_VIDEODRIVER=offscreen CARLA_0.9.12/CarlaUE4.sh -game -opengl -featureleveles31`, per the UE 4.26 documentation to force mobile rendering on desktop, but it exited immediately without meaningful error.

MattRowe18, 2021-11-16:
@bernatx could you comment on this

bernatx, 2021-11-16:
Hi, with the upgrade to UE 4.26 opengl has been deprecated, so only vulkan. You can use -vulkan as a flag, or nothing.

bernatx, 2021-11-16:
Also, to run headless, you need to **unset** the variable SDL_VIDEODRIVER from the environment. At least for us, other definitions of the variable were not working.

closed 2022-04-29 by stale[bot]