carla-simulator/carla #4561: How to perform emergency stop 无痕

EchoYJJ, 2021-08-21:


glopezdiest, 2021-08-30:
Hey @EchoYJJ. According to your comment, "I'm wondering how vehicle can perform emergency stop after collision and then keep static.". This is exactly what you are doing with the collision sensor, you detect a collision, stop, and then when the collision is over, start again. If you want it to remain static my question is, until when? You'd have to add some kind of condition / timer once a collision is detected that will keep the vehicle static.

Regarding agents, what they try to do is predict whether or not a close-by actor is in their route, braking before they collide with it, but this is done using only waypoint lanes and roads, which fail miserably at junctions. We updated the agents for 0.9.12 but that is still an issue we have to resolve. In order to detect collisions, you can just copy the code that implements a collision sensor at *automatic_control.py*, and when a collision occurs, change a class varaible that you are monitoring on runtime.

Did any of this help? Not sure exactly what you are trying to do so I might be talking about something completely unrelated

EchoYJJ, 2021-08-30:
Hi @glopezdiest ,thanks a lot for your reply. It really helps!

closed 2021-08-30 by EchoYJJ